zmkfirmware/zmk #2516: Mod-morph briefly keeps modifier deactivated

lumi-a, 2024-09-29:
Using this behaviour:

```
        debug: debug {
            compatible = "zmk,behavior-mod-morph";
            #binding-cells = <0>;
            bindings = <&kp D>, <&kp B>;
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };
```

All the following works as expected:

- Pressing `&debug`: Produces `d`
- Pressing [shift down], `&debug`, [shift up]: Produces `b`
- Pressing `&debug`, quickly followed by `&kp A`: Produces `da`
- Pressing [shift down], `&debug`, waiting **half a second**, `&kp A`, [shift up]: Produces `bA`

However, unexpectedly:

- Pressing [shift down], `&debug`, **quickly followed by** `&kp A`, [shift up]: Produces `ba`, but I expect this to produce `bA`, as well.

I'm using Github-actions for building. I can reproduce this with both sticky-shift and non-sticky-shift (i.e. `&sk RSHIFT` and `&kp LSHIFT`).

#1898 is a suspiciously similar issue, but was closed by the author because they could no longer reproduce their issue.

lumi-a, 2024-09-29:
Actually, I think the reproduction-steps might be different. It seems it's not the delay between `&debug` and `kp A` that triggers it, but whether `&debug` is still held down when `&kp A` is pressed. It's kind of hard to test which of the two (delay or still-held-down) it is, though. If it's the latter, reproduction-steps become:

- Pressing [shift down], [`&debug` down], [`&debug` up], `&kp A`, [shift up]: Produces `bA` (expected)
- Pressing [shift down], [`&debug` down], `&kp A`, [`&debug` up], [shift up]: Produces `ba` (still unexpected)

Again, these steps also seem to work with sticky-shift.

caksoylar, 2024-09-30:
> The second behaviour is still unexpected, but maybe it's impossible to fix in mod-morphs, because it'd mean I'm asking ZMK to produce [unshifted b] + [shifted A] at the same time. 

I'd think this is the case, since mod-morph waits until you release its key (b here) in order to unmask shift (i.e. press it again).

> If that's the case, is there a ZMK-feature I could use, instead of mod-morphs, for this purpose?

Only mod-morphs do conditional logic based on modifier state, so in the strictest sense there is no other feature to do this. But if you explain what you want to do at a higher level (since I am guessing your D/B example is not it?), maybe some other feature is more appropriate that doesn't depend on modifiers.

lumi-a, 2024-09-30:
Thank you for taking the time to reply!

My particular use-case is a key that produces `$` when non-shift-pressed, and `@` when shift-pressed. So I used this mod-morph:
```
        dollar_at: dollar_at {
            compatible = "zmk,behavior-mod-morph";
            #binding-cells = <0>;
            bindings = <&kp DE_DOLLAR>, <&kp DE_AT_SIGN>;
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };
```
I'm using [`&kp DE_DOLLAR`](https://github.com/joelspadin/zmk-locale-generator/) (as opposed to `&kp DOLLAR`) because my OS is set to a German keyboard layout. However, if I switch my OS to US-QWERTY and use this behaviour:
```
        dollar_at: dollar_at {
            compatible = "zmk,behavior-mod-morph";
            #binding-cells = <0>;
            bindings = <&kp DOLLAR>, <&kp AT_SIGN>;
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };
```
the issue is the same: I'd like to type something like "`@O`" quickly, but have to let go of the Dollar/AtSign-key before pressing the O-key, otherwise I'll produce "`@o`".

Ideas for how to resolve this with other ZMK-features:
- Instead of having a shift-key with mod-morphs, have a whole layer that sends every "normal" key shifted, except for the few mod-morph keys, which send their mod-counterparts on this layer. 
    - This sounds like a generically bad, unidiomatic, non-robust idea.
    - Not sure if it's possible to activate just a shift-press when pressing the layer-key but no other key; in particular, whether "shift"+"mouse-click" could be made to work correctly. Maybe with [layer-taps](https://zmk.dev/docs/keymaps/behaviors/layers#layer-tap)?
- If I used US-QWERTY, I could add `keep-mods = <(MOD_RSFT|MOD_LSFT)>;` to the mod-morph, because the `@` symbol is `shift+2` on US-QWERTY.
    - On QWERTZ, `@` is produced by `altgr+q`, so adding `keep-mods` for QWERTZ would always send `shift+altgr+q` instead. I tried this, and indeed it doesn't produce an `@` anymore.
    - I probably won't switch to US-QWERTY, because I'd like to use special characters of the QWERTZ-layout.
    - Even if someone did use US-QWERTY, they might want a mod-morph not for Dollar/At-Sign, but a mod-morph for Comma/Semicolon, in which case `keep-mods` wouldn't work, since Semicolon is a non-shift-symbol on US-QWERTY.
- I also tried modifying the mod-morph to not send `&kp AT_SIGN`, but to send some macro that has:
    ```
    bindings
    = <&macro_tap &kp AT_SIGN>
    , <&macro_press &kp LSHFT>
    , <&macro_pause_for_release>
    , <&macro_release &kp LSHFT>
    ;
    ```
    so that after the macro sends `@`, shift remains active.
    - This didn't work, the issue persisted, I guess because the mod-morph still masks Shift, as you mentioned.
- The previous two ideas can be combined, though: Define a macro:
    ```
    unshift_at_shift: unshift_at_shift {
        compatible = "zmk,behavior-macro";
        #binding-cells = <0>;
        wait-ms = <1>;
        tap-ms = <1>;
        bindings
        = <&macro_release &kp RSHFT>
        , <&macro_tap &kp DE_AT_SIGN>
        , <&macro_press &kp RSHFT>
        ;
    };
    ```
    This releases left-shift, inputs `@`, and re-presses left-shift. Now rewrite the dollar-at mod-morph to:
    ```
    dollar_at: dollar_at {
        compatible = "zmk,behavior-mod-morph";
        #binding-cells = <0>;
        bindings = <&kp DE_DOLLAR>, <&unshift_at_shift>;
        mods = <(MOD_RSFT)>;
        keep-mods = <(MOD_RSFT)>;
    };
    ```
    It sends `&unshift_at_shift` instead of `&kp DE_AT_SIGN`, and it has `keep-mods = <(MOD_LSFT)>;`.
    - This _works_, but make no mistake: It's an alarmingly bad fix.
    - Adding `wait-ms = <1>; tap-ms = <1>;` was necessary, because otherwise, if I pressed `<shift>@o</shift>` quickly enough, I'd still get `"@o"` instead of `"@O"`. Not sure how robust these super-small wait-times and tap-times are, but they work for now. If non-robust, they could also be increased a little without me accidentally triggering `"@o"`.
    - One consequence is that the `@` key can't be held down anymore. This consequence is _necessary_ to resolve the issue, though, because otherwise I'd be asking for "can you input `shift+o` and `nonshift+@` at the same time, please", aka the impossible.

Is there a better fix than this? Maybe there is some way to write a macro that goes "Do [this] if shift is currently pressed, otherwise do [that]"?

minusfive, 2024-09-30:
What happens if you use `N2` instead of `AT_SIGN` in the US version?

lumi-a, 2024-10-01:
Another possible fix:
- Admit defeat and change your layout, by moving characters that cause issues like this.
    - For my particular QWERTZ-usecase, for instance, I could swap the positions of `$` and `@` (so that `$` is a shift-mod-morph of the `@`-key, instead of the other way around), and add `keep-mods = <(MOD_LSFT|MOD_RSFT)>;` to their mod-morph:
        ```
        at_dollar: at_dollar {
            compatible = "zmk,behavior-mod-morph";
            #binding-cells = <0>;
            bindings = <&kp DE_AT_SIGN>, <&kp DE_DOLLAR>;
            keep-mods = <(MOD_LSFT|MOD_RSFT)>; // !!!
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };
        ```
        This works, because `$` is `shift+5` on QWERTZ, so we can simply unmask the shift.
        
        I could also move the `@` to e.g. my symbol-layer.

lumi-a, 2024-10-01:
(Sorry, accidentally closed due to commit-message in my own keymap-repo)

caksoylar, 2024-10-01:
> Is there a better fix than this? Maybe there is some way to write a macro that goes "Do [this] if shift is currently pressed, otherwise do [that]"

No, macros can't have conditionals like that, other than through combining them with other behaviors (like mod-morph which supplies the conditional). I like that you've figured out a working solution, even if ugly :)

While this is personal opinion territory, I generally believe that using mod-morphs is usually not the right way to achieve alternate functionality like this. Layers are the main mechanism for doing that, and after all, shift is like a layer that the OS applies rather than internally in the keyboard. As a result, there will inevitably be edge cases with mod-morphs, due to having to mask mods etc.

> Instead of having a shift-key with mod-morphs, have a whole layer that sends every "normal" key shifted, except for the few mod-morph keys, which send their mod-counterparts on this layer.
This sounds like a generically bad, unidiomatic, non-robust idea.

Following from above, I'd say I disagree here, I'd even say that for smaller keyboards having a symbol layer (which includes many shifted symbols) is the idiomatic solution. (some samplings at https://keymapdb.com/)

> Not sure if it's possible to activate just a shift-press when pressing the layer-key but no other key; in particular, whether "shift"+"mouse-click" could be made to work correctly. Maybe with [layer-taps](https://zmk.dev/docs/keymaps/behaviors/layers#layer-tap)?

No it isn't possible, at least not easily. But e.g. I don't normally tap symbols and then use mouse to click while holding down the shift key in between, so it hasn't been an issue with me to use a symbol layer and also have a separate shift key (for capitalization, mouse clicks etc.)

While I am aware that I am making subjective statements, I hope the advice is useful.

lumi-a, 2024-10-01:
Thank you for taking the time to reply and for your advice! I also appreciate you giving your subjective opinion. Happy to close this now 🌸

closed 2024-10-01 by lumi-a