zmkfirmware/zmk #2521: setup.sh requires GH repo but gives error that repo exists

vKOALAv, 2024-10-01:
so which is it yall?  the setup explicitly says "please create this in GH first!" which I do exactly that then the script errors out "zmk-config" already exists.  It is created and it is completely empty.

joelspadin, 2024-10-01:
What is the full output of the script? Are you sure it is complaining that the repo exists in GitHub, and not for example that you already have a local folder named "zmk-config" at that location?

vKOALAv, 2024-10-02:
"* GitHub Repo To Push (please create this in GH first!): https://github.com/vKOALAv/zmk-config.git
Continue? [Yn]: Y
fatal: destination path 'zmk-config' already exists and is not an empty directory."

it is an empty repo.

vKOALAv, 2024-10-02:
and if i dont create the repo in GH first:

"* GitHub Repo To Push (please create this in GH first!): https://github.com/vKOALAv/zmk-glove.git
Continue? [Yn]: Y
Cloning into 'zmk-glove'...
remote: Enumerating objects: 52, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 52 (delta 5), reused 4 (delta 4), pack-reused 39 (from 1)
Receiving objects: 100% (52/52), 7.89 KiB | 3.95 MiB/s, done.
Resolving deltas: 100% (10/10), done.
/e/KOALA/kevdev/repositories/zmk/app/zmk-glove/config /e/KOALA/kevdev/repositories/zmk/app/zmk-glove
Downloading config file (https://raw.githubusercontent.com/zmkfirmware/zmk/main/app/boards/arm/glove80/glove80.conf)
Could not find it, falling back to https://raw.githubusercontent.com/zmkfirmware/zmk/main/app/boards/arm/glove80/glove80.conf
Downloading keymap file (https://raw.githubusercontent.com/zmkfirmware/zmk/main/app/boards/arm/glove80/glove80.keymap)
/e/KOALA/kevdev/repositories/zmk/app/zmk-glove
Initialized empty Git repository in E:/KOALA/kevdev/repositories/zmk/app/zmk-glove/.git/
warning: in the working copy of 'build.yaml', LF will be replaced by CRLF the next time Git touches it
warning: in the working copy of 'config/glove80.conf', LF will be replaced by CRLF the next time Git touches it
warning: in the working copy of 'config/glove80.keymap', LF will be replaced by CRLF the next time Git touches it
[master (root-commit) 3391759] Initial User Config.
 7 files changed, 153 insertions(+)
 create mode 100644 .github/workflows/build.yml
 create mode 100644 boards/shields/.gitkeep
 create mode 100644 build.yaml
 create mode 100644 config/glove80.conf
 create mode 100644 config/glove80.keymap
 create mode 100644 config/west.yml
 create mode 100644 zephyr/module.yml
remote: Repository not found.
fatal: repository 'https://github.com/vKOALAv/zmk-glove.git/' not found"

caksoylar, 2024-10-02:
I think you missed Joel's question: 
> Are you sure it is complaining that the repo exists in GitHub, and not for example that you already have a local folder named "zmk-config" at that location?

This error usually happens when you already have a folder called `zmk-config` in your current directory where you run the script. It isn't referring to the remote repo on Github.

vKOALAv, 2024-10-03:
Thanks for the replies yall. I will check it out tonight. Been a couple long work days at the office

vKOALAv, 2024-10-05:
Thanks again.  I shouldn't have started this project in the middle of that week.  I couldn't even find where I had started this on my local drives.  So I started over and I didn't have the problem.  I guess I have the local zmk-config somewhere.

closed 2024-10-05 by vKOALAv