zmkfirmware/zmk #2526: fix: `keydown` and `keyup` event occurs simultaneously when holding `&mo`

jjangga0214, 2024-10-02:
Hi!

Let's say I have a keymap like this.

```
#define LAYER_B 1

/ {
    behaviors {
        hlt: hold_preferred_lt {
            compatible = "zmk,behavior-hold-tap";
            label = "hold_preferred_lt";
            #binding-cells = <2>;
            tapping-term-ms = <160>;
            flavor = "hold-preferred";
            bindings = <&mo>, <&kp>;
        };
    };

    keymap {
        compatible = "zmk,keymap"; 
        a { 
            &kp A             &hlt LAYER_B C
        };
        b { 
            &kp LCMD     &trans
        };       
    };    
};
```

There're two physical keys.
When I hold the second key, 

### Steps for what I expected

0. Layer A is active by default.
1.  Keep Holding the second key.
2. `&mo LAYER_B` takes effect.
3. Hold the first key.
4. `keydown` event for `LCMD` occurs.
5. Stop holding the first key.
6. `keyup` event for `LCMD` occurs.
7. Stop holding the second key.
8. Layer A is activated again.

### Steps for what actually happens

0. Layer A is active by default.
1.  Keep Holding the second key.
2. `&mo LAYER_B` takes effect.
3. Hold the first key.
4. **No event occurs.**
5. Stop holding the first key.
6. **`keydown` and `keyup` event for `LCMD` occurs.**
7. Stop holding the second key.
8. Layer A is activated again.

Step 4 and 6 are different from what I expected.
Is this an intended behavior?

caksoylar, 2024-10-02:
Yes, I'd say this is intended. When you press a key, it acts like the binding you press at that specific layer until you release that key itself.

Off the top of my head, this has the main benefit of being able to put modifiers on a layer, then while holding them being able to press keys on the base layer to do shortcuts. 

It also removes any potential requirement of duplicating &mo keys on the layer they are going to, where otherwise it either wouldn't work or the layer would get stuck on.

caksoylar, 2024-10-02:
This is not normal, but I don't see anything wrong with the keymap either. It's also not something I can reproduce on my keymap which has a similar setup. Maybe you aren't trying with exactly this keymap? If you are, you might want to reproduce it with a [test](https://zmk.dev/docs/development/local-toolchain/tests#creating-a-new-test-set).

jjangga0214, 2024-10-03:
I am very new to west.
Should the test be like this?

```
#include "../behavior_keymap.dtsi"

&kscan {
    events = <
        ZMK_MOCK_PRESS(0,1,1000)
        ZMK_MOCK_PRESS(0,0,1000)
        ZMK_MOCK_RELEASE(0,0,1000)
        ZMK_MOCK_RELEASE(0,1,1000)
    >;
};
```

jjangga0214, 2024-10-03:
By the way the original config is this.

https://github.com/jjangga0214/zmk-config-adv360-pro/blob/674953b43e7c454254601d40c1cae0a10c92859e/config/adv360.keymap#L707-L725

[image]

caksoylar, 2024-10-05:
That LCMD position seems to have 18 combos, with tapping terms up to 105ms. I don't know for sure that it is related but maybe it is, it might be worth trying to isolate the issue with a simpler setup.

jjangga0214, 2024-10-06:
@caksoylar I found it's not reproducible. 

**Sometimes it works well, but some other times it doesn't!** 

Its behavior randomly changes.

jjangga0214, 2024-10-07:
I found rebooting the keyboard temporarily solves the issue.

1. The issue happens.
2. Reboot the keyboard
3. The issue is resolved temporarily.
4 After a few day, the issue happens again.
 
Maybe a memory leak or bug on state logic?