zmkfirmware/zmk #2534: Better Documentation around configs

SamBobBarnes, 2024-10-04:
I ran into an issue recently that could have been mitigated with some teaks to the documentation.
Specifically, [https://zmk.dev/docs/config](https://zmk.dev/docs/config). It would have helped a ton to have an example file structure for a minimal config. I was tripped up because I didnt understand that the `my-keyboard.conf` file needed to belong in the config folder rather than the `shield/my-keyboard` folder.

caksoylar, 2024-10-04:
Where did you initially come across adding that file? I am asking because although we have it documented in a few places, it is still a frequent footgun for users.

For reference, the config page mentions:

> Shared config files (excluding any _left or _right suffix) are not currently supported in shield folders.

New shield docs have this note (if you select the split tab):

> The shared configuration in my_keyboard.conf is only applied when you are building with a [zmk-config folder](https://zmk.dev/docs/development/local-toolchain/build-flash#building-from-zmk-config-folder) and it is present at config/my_keyboard.conf.

I'd say that maybe moving it to "Configuration elements affect both halves" would be better.

> It would have helped a ton to have an example file structure for a minimal config.

This is a good idea, I can see that helping in https://zmk.dev/docs/development/hardware-integration/new-shield#new-shield-directory.

For the Configuration Overview page, maybe adding tabs for non-split/split (and/or ZMK/config/module) would make it clearer.

caksoylar, 2025-01-06:
Closing since #2634 should help.

closed 2025-01-06 by caksoylar