zmkfirmware/zmk #2535: West Build Error: parse error: expected number or parenthesized expression

Da4ndo, 2024-10-04:
Hello, I encountered a problem while building my configuration. It says there is a parse error on line 92, but I notice nothing wrong there or anywhere else.
Also, please feel free to provide suggestions for improving my keymap.

**Error:** `devicetree error: /tmp/zmk-config/config/corne.keymap:92 (column 164): parse error: expected number or parenthesized expression` <--

My config: https://github.com/Da4ndo/zmk-config-thoth

### West Build Log
```
Run west build -s zmk/app -d "/tmp/tmp.WPVzFJz59p" -b "nice_nano_v2"  -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD="corne_left nice_view_adapter nice_view" -DZMK_EXTRA_MODULES='/__w/zmk-config-thoth/zmk-config-thoth' 
+ west build -s zmk/app -d /tmp/tmp.WPVzFJz59p -b nice_nano_v2 -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD=corne_left nice_view_adapter nice_view -DZMK_EXTRA_MODULES=/__w/zmk-config-thoth/zmk-config-thoth
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /tmp/zmk-config/zmk/app
-- CMake version: 3.30.0
-- Found Python3: /usr/bin/python3 (found suitable version "3.12.3", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /tmp/zmk-config/zephyr/.cache
-- Zephyr version: 3.5.0 (/tmp/zmk-config/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Adding /tmp/zmk-config/zmk/app/boards/shields/corne
-- Adding /tmp/zmk-config/zmk/app/boards/shields/nice_view_adapter
-- Adding /tmp/zmk-config/zmk/app/boards/shields/nice_view
-- ZMK Config directory: /tmp/zmk-config/config
-- ZMK Config Kconfig: /tmp/zmk-config/config/corne.conf
-- Using keymap file: /tmp/zmk-config/config/corne.keymap
-- Using keymap file: /tmp/zmk-config/config/corne.keymap
-- Board: nice_nano_v2
-- Shield(s): corne_left nice_view_adapter nice_view
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found Dtc: /opt/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /tmp/zmk-config/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /tmp/zmk-config/zmk/app/boards/shields/corne/corne_left.overlay
-- Found devicetree overlay: /tmp/zmk-config/zmk/app/boards/shields/corne/boards/nice_nano_v2.overlay
-- Found devicetree overlay: /tmp/zmk-config/zmk/app/boards/shields/nice_view_adapter/nice_view_adapter.overlay
-- Found devicetree overlay: /tmp/zmk-config/zmk/app/boards/shields/nice_view_adapter/boards/nice_nano_v2.overlay
-- Found devicetree overlay: /tmp/zmk-config/zmk/app/boards/shields/nice_view/nice_view.overlay
-- Found devicetree overlay: /tmp/zmk-config/config/corne.keymap
devicetree error: /tmp/zmk-config/config/corne.keymap:92 (column 164): parse error: expected number or parenthesized expression
CMake Error at /tmp/zmk-config/zephyr/cmake/modules/dts.cmake:279 (message):
-- In: /tmp/tmp.WPVzFJz59p/zephyr, command: /usr/bin/python3;/tmp/zmk-config/zephyr/scripts/dts/gen_defines.py;--dts;/tmp/tmp.WPVzFJz59p/zephyr/zephyr.dts.pre;--dtc-flags;'-Wno-unique_unit_address_if_enabled';--bindings-dirs;/tmp/zmk-config/zmk/app/module/dts/bindings;/tmp/zmk-config/zmk/app/dts/bindings;/tmp/zmk-config/zephyr/dts/bindings;--header-out;/tmp/tmp.WPVzFJz59p/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/tmp/tmp.WPVzFJz59p/zephyr/zephyr.dts.new;--edt-pickle-out;/tmp/tmp.WPVzFJz59p/zephyr/edt.pickle;--vendor-prefixes;/tmp/zmk-config/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/tmp/zmk-config/zephyr/dts/bindings/vendor-prefixes.txt
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /tmp/zmk-config/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)
-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/tmp.WPVzFJz59p -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/tmp/zmk-config/config '-DSHIELD=corne_left nice_view_adapter nice_view' -DZMK_EXTRA_MODULES=/__w/zmk-config-thoth/zmk-config-thoth -S/tmp/zmk-config/zmk/app
Error: Process completed with exit code 1.
```

### Keymap

(92 line is adjust layer`s second key row. Also there is no column 164, it only goes to like 130 or so)

```
/*
 * Copyright (c) 2020 The ZMK Contributors
 *
 * SPDX-License-Identifier: MIT
 */

#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/outputs.h>
#include <dt-bindings/zmk/mouse.h>

[... 64 lines cut ...]
                &kp F6  &kp F7  &kp F8  &kp F9  &kp F10    &none     &none            &kp UP    &none     &kp PG_UP
                &kp F11 &kp F12 &none   &none   &none      &none     &kp LEFT         &kp DOWN  &kp RIGHT &kp PG_DN
                               &kp LGUI &tog ADJUST &kp SPACE   AS(RET)   &none     &kp TAB
            >;
        };
        
        adjust_layer {
            // -----------------------------------------------------------------------------------------
            // | PREV | PLAY | NEXT | MUTE | VOL+ |   |     |     |     |     |     |
            // | BRI+ | BRI- |      |      | VOL- |   |BTCLR| LCLK| MUP | RCLK| SCUP|
            // | BT0  | BT1  | BT2  | BT3  | BT4  |   |     |MLEFT|MDOWN|MRGHT| SCDN|
            //                | GUI | BASE | SPC  |   | ENT |     | TAB |
            display-name = "Adjust";
            bindings = <
                &kp C_PREV    &kp C_PP      &kp C_NEXT    &kp C_MUTE    &kp C_VOL_UP    &none       &none           &none           &none           &none
                &kp C_BRI_UP  &kp C_BRI_DN  &none         &none         &kp C_VOL_DN    &bt BT_CLR  &mkp LCLK       &mmv MOVE_UP    &mkp RCLK       &mwh SCROLL_UP
                &bt BT_SEL 0  &bt BT_SEL 1  &bt BT_SEL 2  &bt BT_SEL 3  &bt BT_SEL 4    &none       &mmv MOVE_LEFT  &mmv MOVE_DOWN  &mmv MOVE_RIGHT &mwh SCROLL_DOWN
                                            &kp LGUI      &to DEFAULT   &kp SPACE       &kp RET     &none           &kp TAB
            >;
        };

        
    };
};
```

joelspadin, 2024-10-04:
See https://zmk.dev/docs/troubleshooting/building-issues for how to interpret that error.

There is no &mwh behavior in ZMK's main branch. If you are trying to use features from a different branch, you need to fix your west.yml.

Da4ndo, 2024-10-04:
Removed it, but now I have this error:
```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /tmp/tmp.V4foKoV0WK/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/tmp/tmp.V4foKoV0WK/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/soc/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/bluetooth/crypto/libsubsys__bluetooth__crypto.a  zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a  zephyr/subsys/net/libsubsys__net.a  zephyr/drivers/usb/device/libdrivers__usb__device.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/display/libdrivers__display.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/i2c/libdrivers__i2c.a  zephyr/drivers/kscan/libdrivers__kscan.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a  zephyr/drivers/spi/libdrivers__spi.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/lvgl/libmodules__lvgl.a  modules/module/drivers/sensor/battery/lib..__zmk__app__module__drivers__sensor__battery.a  modules/module/lib/zmk_debounce/lib..__zmk__app__module__lib__zmk_debounce.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m+fp/hard"  -L/tmp/tmp.V4foKoV0WK/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfpu=fpv4-sp-d16  -mfloat-abi=hard  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_INTEGER_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /tmp/tmp.V4foKoV0WK/zephyr && /usr/local/lib/python3.12/dist-packages/cmake/data/bin/cmake -E true
/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(peripheral_status.c.obj):/tmp/zmk-config/zmk/app/boards/shields/nice_view/widgets/peripheral_status.c:82: multiple definition of `widget_battery_status_mutex'; app/libapp.a(battery_status.c.obj):/tmp/zmk-config/zmk/app/src/display/widgets/battery_status.c:76: first defined here
/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(peripheral_status.c.obj):/tmp/zmk-config/zmk/app/boards/shields/nice_view/widgets/peripheral_status.c:82: multiple definition of `widget_battery_status_work'; app/libapp.a(battery_status.c.obj):/tmp/zmk-config/zmk/app/src/display/widgets/battery_status.c:76: first defined here
/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(peripheral_status.c.obj):/tmp/zmk-config/zmk/app/boards/shields/nice_view/widgets/peripheral_status.c:87: multiple definition of `zmk_event_sub_widget_battery_statuszmk_usb_conn_state_changed'; app/libapp.a(battery_status.c.obj):/tmp/zmk-config/zmk/app/src/display/widgets/battery_status.c:81: first defined here
/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(peripheral_status.c.obj):/tmp/zmk-config/zmk/app/boards/shields/nice_view/widgets/peripheral_status.c:82: multiple definition of `zmk_listener_widget_battery_status'; app/libapp.a(battery_status.c.obj):/tmp/zmk-config/zmk/app/src/display/widgets/battery_status.c:76: first defined here
/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/libzephyr.a(peripheral_status.c.obj):/tmp/zmk-config/zmk/app/boards/shields/nice_view/widgets/peripheral_status.c:85: multiple definition of `zmk_event_sub_widget_battery_statuszmk_battery_state_changed'; app/libapp.a(battery_status.c.obj):/tmp/zmk-config/zmk/app/src/display/widgets/battery_status.c:79: first defined here
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /tmp/tmp.V4foKoV0WK
Error: Process completed with exit code 1.
```

joelspadin, 2024-10-04:
You are trying to enable the default status screen's battery widget https://github.com/Da4ndo/zmk-config-thoth/blob/master/config/corne.conf#L6, but you are using the nice_view shield which has its own screen and battery widget.

Da4ndo, 2024-10-04:
It built successfully after the fix. Thank you for your help.

closed 2024-10-04 by Da4ndo

joelspadin, 2024-10-04:
You're welcome!