zmkfirmware/zmk #2536: Missing physical layouts

Nick-Munnich, 2024-10-05:
This is a checklist of all the physical layouts that still need to be defined for in-tree keyboards.

Boards:

- [ ] s40nc
- [ ] dz60rgb

Shields:

- [ ] a-dux
- [ ] bat43
- [ ] bfo9000
- [ ] boardsource 3x4 (also missing matrix transform)
- [ ] chalice
- [ ] clog
- [ ] clueboard_california
- [ ] eek
- [ ] elephant42
- [ ] ergodash
- [ ] eternal_keypad
- [ ] fourier
- [ ] helix (shared layout?)
- [ ] iris (shared layout?)
- [ ] jian
- [ ] jiran
- [ ] jorne
- [ ] knob_goblin
- [ ] leeloo
- [ ] leeloo_micro
- [ ] lotus58
- [ ] microdox
- [ ] murphpad
- [ ] naked60 (also missing matrix transform)
- [ ] nibble
- [ ] osprette
- [ ] pancake (also missing matrix transform)
- [ ] qaz
- [ ] quefrency
- [ ] redox
- [ ] reviung34
- [ ] reviung5
- [ ] reviung53
- [ ] romac
- [ ] romac_plus
- [ ] snap
- [ ] aurora_helix (shared layout?)
- [ ] splitreus62
- [ ] tg4x
- [ ] tidbit
- [ ] two_percent_milk
- [ ] waterfowl
- [ ] zodiark

Please state below which ones you're writing if you want to help out, to prevent anyone else from wasting their time. Opinions on whether any of these should be given shared layouts can also be written below.

krikun98, 2024-10-07:
https://github.com/zmkfirmware/zmk/pull/2538 should handle the Jian, Jiran and Jorne.

brmejia, 2025-09-14:
Added Redox layout #3068