zmkfirmware/zmk #2540: Layer numbers absent from nice!view on Typeractive Corne and Lily58

amacleod, 2024-10-07:
See also https://github.com/zmkfirmware/zmk/issues/2467 and https://github.com/zmkfirmware/zmk/pull/2491

When using layers with no `display-name`, the widget at the very bottom of the left screen is totally blank. My expectation is that the numeric index of the layer would be printed there. I have reproduced this on my Typeractive Lily58 Choc wireless, and another person on the Typeractive Discord reported this behavior on their Typeractive Corne wireless.

It is possible to work around this by adding `display-name` values to every layer.

amacleod, 2024-10-07:
The code with which I tested is https://github.com/amacleod/zmk-config/tree/557276552c30d3fd1ecc2638cebcc13c15fa349a on 2024-10-07 at which time the latest commit on ZMK main was `29b39f2b9f0977098601b201be1942d299b44da5`

caksoylar, 2024-10-07:
I thought #2491 fixed this, good catch.

closed 2024-10-08 by petejohanson