zmkfirmware/zmk #2550: Where to put drivers?

kaievns, 2024-10-11:
Hi folks, I need a bit of an advise on where to put drivers in my config.

This is not my first zmk project and I'm trying something more ambitious this time with a new custom  PCB keyboard build that has a trackball integrated.

I have set up a new shield and MCU overlay for the new PCB, transformations, keybindings, etc, and it compiles and works. Now I'm trying to add the trackball driver to the thing.

Basically what I want to accomplish is to copy and integrate this project https://github.com/inorichi/zmk-pmw3610-driver into my configs so I could tinker with the code locally without dealing with the full blown github actions, west, etc.

I have stashed my work over here, it's pretty vanilla, I just don't know where all the C and Kconfig files from the driver should go in here https://github.com/kaievns/little-wing/tree/main/zmk-config

Thank you!

kaievns, 2024-10-11:
I think I found a way to use the extra modules ref for that.

I have basically dumped that repo into an arbitrary folder `${my-config}/drivers/pmw3610` and then referred to that as an external zephyr module in the CLI args like so

```
west build -d build/right -p -b nice_nano_v2 -- -DSHIELD="little_wing_right" -DZMK_CONFIG="/workspaces/zmk-config/config" -DZMK_EXTRA_MODULES="/workspaces/zmk-config;/workspaces/zmk-config/drivers/pmw3610"
```

the `/workspaces/zmk-config` is my docker mount for my user config.

it works, but it sort of feels hacky. is there a more officially blessed way to do this?

thank you

caksoylar, 2024-10-12:
You found the right way, they are "modules" and hence when building locally (where the `config/west.yml` manifest isn't used) you have to use the `-DZMK_EXTRA_MODULES` arg to pass them to the build command. This is the official way that is documented at https://zmk.dev/docs/development/local-toolchain/build-flash#building-with-external-modules.

Nick-Munnich, 2024-10-12:
To expand on the above answer, you could also adjust the west configuration to prevent you from needing to input those values every time (though I don't know about the interaction with docker at all). See https://docs.zephyrproject.org/3.5.0/develop/west/build-flash-debug.html#permanent-cmake-arguments

kaievns, 2024-10-13:
thanks folks, I'm closing this then

closed 2024-10-13 by kaievns