zmkfirmware/zmk #2551: Check for combos exceeding CONFIG_ZMK_COMBO_MAX_COMBOS_PER_KEY at compile time.

amacleod, 2024-10-13:
As a quality of life feature, it would be nice to have a way to automatically detect excess combos and issue a compile time warning. Or possibly even use that detection to implicitly increase the combo-per-key limit within reason.

Currently, if I overload a key with more combos than `CONFIG_ZMK_COMBO_MAX_COMBOS_PER_KEY` allows, it seems I silently lose the use of one or more of those combos. Rather than discovering this when I try to use that combo, I would prefer to have an error or warning when I (or the GH Action) go to build the firmware.

amacleod, 2024-10-13:
To be clear, I consider this an extremely minor nice-to-have. At some point I might take a crack at a PR for it, but only after learning more about ZMK internals 😄

joelspadin, 2024-10-13:
If it's possible to calculate the number of combos using each key at compile time in order to check if it exceeds the max, then it should also be possible to just calculate what the max should be and eliminate the need for the setting entirely.

urob, 2024-10-13:
Could one overload `west build` with a custom west command that's a shallow Python wrapper around the original one?

One could run the config through the preprocessor first and then use python to parse and automate various settings. 

One might also do the same to improve error messages. E.g., showing the code corresponding to problematic columns after all macros have been expanded.

amacleod, 2025-02-28:
I think this issue would be solved (or at least obviated) by https://github.com/zmkfirmware/zmk/pull/2849 if I am reading PR 2849 right.