zmkfirmware/zmk #2552: MacOS: ModuleNotFoundError: No module named 'elftools'

sommerper, 2024-10-13:
When trying to build a shield I get the error
```
ModuleNotFoundError: No module named 'elftools'
```

```
zmk/app on  main is 📦 v1.0.0 via △ v3.30.2 via  v22.5.1
❯ west build -p -b nrfmicro_13 -- -DSHIELD=corne_left
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /Users/ps/development/github/zmk/app
-- CMake version: 3.30.2
-- Found Python3: /Users/ps/.local/pipx/venvs/west/bin/python (found suitable version "3.12.5", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /Users/ps/Library/Caches/zephyr
-- Zephyr version: 3.5.0 (/Users/ps/development/github/zmk/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Adding /Users/ps/development/github/zmk/app/boards/shields/corne
-- Using keymap file: /Users/ps/development/github/zmk/app/boards/shields/corne/corne.keymap
-- Board: nrfmicro_13
-- Shield(s): corne_left
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.8 (/Users/ps/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/Users/ps/zephyr-sdk-0.16.8)
-- Could NOT find Dtc (missing: DTC) (Required is at least version "1.4.6")
Hint: The project() command has not yet been called.  It sets up system-specific search paths.
-- Found BOARD.dts: /Users/ps/development/github/zmk/app/boards/arm/nrfmicro/nrfmicro_13.dts
-- Found devicetree overlay: /Users/ps/development/github/zmk/app/boards/shields/corne/corne_left.overlay
-- Found devicetree overlay: /Users/ps/development/github/zmk/app/boards/shields/corne/corne.keymap
-- Generated zephyr.dts: /Users/ps/development/github/zmk/app/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /Users/ps/development/github/zmk/app/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /Users/ps/development/github/zmk/app/build/zephyr/dts.cmake

warning: Deprecated symbol NRF_STORE_REBOOT_TYPE_GPREGRET is enabled.

Parsing /Users/ps/development/github/zmk/app/Kconfig
Loaded configuration '/Users/ps/development/github/zmk/app/boards/arm/nrfmicro/nrfmicro_13_defconfig'
Merged configuration '/Users/ps/development/github/zmk/app/prj.conf'
Merged configuration '/Users/ps/development/github/zmk/app/boards/shields/corne/corne_left.conf'
Configuration saved to '/Users/ps/development/github/zmk/app/build/zephyr/.config'
Kconfig header saved to '/Users/ps/development/github/zmk/app/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /Users/ps/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/ps/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done (2.6s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/ps/development/github/zmk/app/build
-- west build: building application
[1/316] Preparing syscall dependency handling

[2/316] Generating include/generated/version.h
-- Zephyr version: 3.5.0 (/Users/ps/development/github/zmk/zephyr), build: f8e4d1579160
[5/316] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
FAILED: zephyr/include/generated/kobj-types-enum.h zephyr/include/generated/otype-to-str.h zephyr/include/generated/otype-to-size.h /Users/ps/development/github/zmk/app/build/zephyr/include/generated/kobj-types-enum.h /Users/ps/development/github/zmk/app/build/zephyr/include/generated/otype-to-str.h /Users/ps/development/github/zmk/app/build/zephyr/include/generated/otype-to-size.h
cd /Users/ps/development/github/zmk/app/build/zephyr && /Users/ps/.local/pipx/venvs/west/bin/python /Users/ps/development/github/zmk/zephyr/scripts/build/gen_kobject_list.py --kobj-types-output /Users/ps/development/github/zmk/app/build/zephyr/include/generated/kobj-types-enum.h --kobj-otype-output /Users/ps/development/github/zmk/app/build/zephyr/include/generated/otype-to-str.h --kobj-size-output /Users/ps/development/github/zmk/app/build/zephyr/include/generated/otype-to-size.h --include-subsystem-list /Users/ps/development/github/zmk/app/build/zephyr/misc/generated/struct_tags.json
Traceback (most recent call last):
  File "/Users/ps/development/github/zmk/zephyr/scripts/build/gen_kobject_list.py", line 62, in <module>
    import elftools
ModuleNotFoundError: No module named 'elftools'
[6/316] Generating include/generated/driver-validation.h
FAILED: zephyr/include/generated/driver-validation.h /Users/ps/development/github/zmk/app/build/zephyr/include/generated/driver-validation.h
cd /Users/ps/development/github/zmk/app/build/zephyr && /Users/ps/.local/pipx/venvs/west/bin/python /Users/ps/development/github/zmk/zephyr/scripts/build/gen_kobject_list.py --validation-output /Users/ps/development/github/zmk/app/build/zephyr/include/generated/driver-validation.h --include-subsystem-list /Users/ps/development/github/zmk/app/build/zephyr/misc/generated/struct_tags.json
Traceback (most recent call last):
  File "/Users/ps/development/github/zmk/zephyr/scripts/build/gen_kobject_list.py", line 62, in <module>
    import elftools
ModuleNotFoundError: No module named 'elftools'
[7/316] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake --build /Users/ps/development/github/zmk/app/build
```

I should already have the module installed since when running the commands below they state it's already there.

```
❯ pip3 install pyelftools
Requirement already satisfied: pyelftools in /opt/homebrew/lib/python3.12/site-packages (0.31)

❯ pipx install pyelftools
'pyelftools' already seems to be installed. Not modifying existing installation in '/Users/ps/.local/pipx/venvs/pyelftools'. Pass '--force' to force installation.
```

Could there be something I still need configure somewhere?

Thanks!

joelspadin, 2024-10-13:
How did you install West? If you used pipx to install it, then West has its own separate venv which probably does not have pyelftools in it.

Make sure you follow all of the instructions at https://zmk.dev/docs/development/local-toolchain/setup/native

sommerper, 2024-10-13:
I should add to this that I just tried on Asahi Fedora and got to the same error.

I've never had any trouble installing this on PopOS/Ubuntu. There must be something I'm missing.

sommerper, 2024-10-13:
@joelspadin Thanks, it was indeed the pipx issue! I did follow the instructions and the first time I got an error, which is why I ended up using pipx. Now when I tried `pip3 install -U west` I got no complaints.

Thanks a bunch - much appreciated! And sorry for taking your time.

closed 2024-10-13 by sommerper