zmkfirmware/zmk #2560: &kp bootloader stopped working

masar3141, 2024-10-16:
Hello, my keyboard (nrf52) suddenly stopped going into bootloader on keypress (&kp bootloader). 
When i press the key the keyboard stops sending key for few seconds (nothing when i ls /Volumes) then goes back to normal.

I'm on macos, and have in general many issues with zmk.

Any idea please?

eboy79, 2025-01-05:
i believe my nine!nanoV2 clone is doing the same thing...

alexander-neumann-webdesign, 2025-01-11:
I’m experiencing the same issue with my device. I’m using a SuperMini nRF52840 from AliExpress, which I flashed as a ⁠`nice_nano_v2`. The device works well in most aspects, except that entering the bootloader using `⁠&bootloader` doesn’t function correctly. Instead of entering bootloader mode, the device reboots and loads the existing firmware, similar to the behavior of `⁠&sys_reset`.

ssbb, 2025-01-11:
@masar3141 As a side note - you should not use `&kp` with bootloader behavior but reference it directly like `&bootloader`. Not sure it's a typo here or the cause of an issue.

masar3141, 2025-02-06:
> [@masar3141](https://github.com/masar3141) As a side note - you should not use `&kp` with bootloader behavior but reference it directly like `&bootloader`. Not sure it's a typo here or the cause of an issue.

You're right thank you for noting, it was a typo indeed