zmkfirmware/zmk #2573: Build Failed: undefined reference to `zmk_matrix_transform_default'

iBot1986, 2024-10-19:
Hi,
I had a minor change in my keymap and wanted to recompile zmk for my keyboard:
[https://github.com/iBot1986/mechanica-module](https://github.com/iBot1986/mechanica-module)
But the workflow failed with the following error:
/opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(physical_layouts.c.obj):(.rodata._zmk_physical_layout_chosen+0x4): undefined reference to `zmk_matrix_transform_default'
--> See full log in attachment.
[logs_29730936362.zip](https://github.com/user-attachments/files/17446902/logs_29730936362.zip)

The last successful build was 4 months ago; but even with this config, the build fails.
What could be the problem?
I have found a simliar problem here (not mine):
[https://stackoverflow.com/questions/79096192/zmk-and-zephyr-problems-when-compiling](https://stackoverflow.com/questions/79096192/zmk-and-zephyr-problems-when-compiling)

Thanks and best regards,
iBot

caksoylar, 2024-10-19:
You'll want to specify a chosen matrix transform, e.g. add `zmk,matrix-transform = &default_transform;` under line 5 in your [overlay](https://github.com/iBot1986/mechanica-module/blob/main/boards/shields/mechanica/mechanica.overlay#L5).

While you defined the matrix transform it wasn't used before as it wasn't set as chosen. There was a fallback mechanism where it assumed a mapping (and this worked for you due to the full matrix) but this was recently removed in favor of making the matrix transform mandatory.

We now recommend you define a physical layout node in https://zmk.dev/docs/development/hardware-integration/new-shield#physical-layout. While not having it will probably not break anytime soon, you can switch to the new approach if e.g. you want [ZMK Studio](https://zmk.dev/docs/features/studio) support.

iBot1986, 2024-10-20:
Hi caksoylar,
Thanks, that worked very well! The keyboard is building again.
I'll take a look into the physical layout; I'll have to read me in. Also, ZMK Studio sounds like a big improvement!
This ticket can be closed.
Thanks and best regards,
iBot

closed 2024-10-20 by Nick-Munnich