zmkfirmware/zmk #2577: Caps word with tap-dance + hold-tap

salamantos, 2024-10-21:
I'm trying to achieve the following behavior: on single tap it should produce `F`, on hold it should produce `SHIFT`, on double tap and hold (tap-hold) I want it to activate `caps_word`

However, with my current configuration, this is not working as expected. Instead, double tap and hold produce `O` for some reason. The single tap and single hold are working as intended.

```
/ {
    behaviors {
        hrm: home_row_modifier {
            compatible = "zmk,behavior-hold-tap";
            #binding-cells = <2>;
            flavor = "balanced";
            tapping-term-ms = <300>;
            quick-tap-ms = <0>;
            bindings = <&kp>, <&kp>;
        };
        
        hold_caps_word: hold_caps_word {
            compatible = "zmk,behavior-hold-tap";
            #binding-cells = <2>;
            flavor = "tap-preferred";
            tapping-term-ms = <300>;
            quick-tap-ms = <0>;
            bindings = <&kp>, <&kp>;
        };

        f_shft_caps: F_hold_SHIFT_doubletap_CAPS_WORD {
            compatible = "zmk,behavior-tap-dance";
            #binding-cells = <0>;
            tapping-term-ms = <300>;
            bindings = <&hrm LSHFT F>, <&hold_caps_word &caps_word F>;
        };
    };

    keymap {
        compatible = "zmk,keymap";
        default_layer {
            bindings = <
                ...    &f_shft_caps    ...
            >;
        };
    };
};
```

caksoylar, 2024-10-21:
You cannot use other behavior references as parameters to a behavior: `&hold_caps_word &caps_word F` will result in a pseudorandom behavior because the `&caps_word` will be interpreted as a keycode. This is because you used `bindings = <&kp>, <&kp>;` in the `hold_caps_word` definition.

Instead, you need to use `bindings = <&caps_word>, <&kp>;` in the behavior definition, then use a dummy parameter value like `0` when using it: `&hold_caps_word 0 F`. This is documented here: https://zmk.dev/docs/keymaps/behaviors/hold-tap#using-different-behavior-types-with-hold-taps

salamantos, 2024-10-22:
Thanks! It helped, now everything is working perfectly

closed 2024-10-22 by salamantos

Antebios, 2025-01-03:
> Thanks! It helped, now everything is working perfectly

Did you get double tap working, or tap AND THEN HOLD working.  I have single tap, single hold, and double tap working, but I want to get double hold (or tap and then hold, or whatever you call it when you tap once, then release, then press down a second time but this time hold the key down).  I got it working in QMK, but no in ZMK.