zmkfirmware/zmk #2592: Cannot get &lt keys to repeat the tap keycode when tapped and held

dimitrioschr, 2024-10-25:
My keymap uses some &lt behaviours that are related to SPACE, BACKSPACE and DEL and I would like these keys to repeat when the &lt key is held. I have tried a variety of re-arrangements and additions to my keymap but the required output cannot be achieved, I need to press the keys repeatedly to delete multiple characters or add many spaces. Any ideas? Here is my keymap:

`/*
 * Copyright (c) 2020 The ZMK Contributors
 *
 * SPDX-License-Identifier: MIT
 */

#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>

#define _QWERTY   0
#define _NAVI     1
#define _NUMBER   2
#define _FUNCTION 3
#define _SYMBOL   4
#define _BLUE     5

#define dQ &ht EXCL Q
#define dW &ht AT W
#define dE &ht HASH E
#define dR &ht DOLLAR R
#define dT &ht PERCENT T

#define dA &ht LGUI A
#define dS &ht LALT S
#define dD &ht LCTRL D
#define dF &ht LSHIFT F
#define dG &ht PLUS G

#define dZ &kp Z
#define dX &ht LBRC X
#define dC &ht LBKT C
#define dV &ht LPAR V
#define dB &ht ASTERISK B

#define dTILD &ht GRAVE TILDE

#define dY &ht CARET Y
#define dU &ht AMPS U
#define dI &ht ASTERISK I
#define dO &ht UNDERSCORE O
#define dP &ht EQUAL P

#define dH &ht MINUS H
#define dJ &ht LSHIFT J
#define dK &ht LCTRL K
#define dL &ht LALT L
#define dCOLN &ht SEMICOLON COLON

#define dN &ht SLASH N
#define dM &ht RPAR M
#define dCOMM &ht RBKT COMMA
#define dDOT &ht RBRC DOT
#define dSLSH &ht QUESTION SLASH

#define dDQT &ht SQT DQT
#define dBSLS &ht PIPE BACKSLASH

#define dOSL &sl _SYMBOL

#define dBSPC &lt _NUMBER BSPC
#define dDEL &lt _FUNCTION DEL
#define dSPC &lt _NAVI SPACE

/* REMEMBER TO DEFINE THE &ht HOLD TAP BEHAVIOUR*/

/ {

[... 182 lines cut ...]
                bluetooth_layer {
/* .----------------------------------------------------------------------------------------------------------------------.
 * |      |      |      |      |      |      |      |                    |      |      |      |      |      |      |      |
 * |------+------+------+------+------+------+------+--------------------+------+------+------+------+------+------+------|
 * |      |      |      |      |      |      |      |                    |      |      |SEL(0)|SEL(1)|SEL(2)|      |      |
 * |------+------+------+------+------+------+------+--------------------+------+------+------+------+------+------+------|
 * |      |      |      |      |      |      |      |                    |      |      |DSC(0)|DSC(1)|DSC(2)|      |      |
 * |------+------+------+------+------+------+---------------------------+------+------+------+------+------+------+------|
 * |      |      |      |      |      |      |      |                    |      |      |  CLR |  CLR |  CLR |      |      |
 * |-------------+------+------+------+------+------+------+------+------+------+------+------+------+------+-------------|
 * |      |      |      |      ||||||||      |      |      ||||||||      |      |      ||||||||      |      |      |      |
 * .----------------------------------------------------------------------------------------------------------------------.
 */
                        bindings = <
&none         &none         &none         &none         &none         &none         &none                  &none         &none         &none         &none         &none         &none         &none
&none         &none         &none         &none         &none         &none         &none                  &none         &none         &bt BT_SEL 0  &bt BT_SEL 1  &bt BT_SEL 2  &none         &none
&none         &none         &none         &none         &none         &none         &none                  &none         &none         &bt BT_DISC 0 &bt BT_DISC 1 &bt BT_DISC 2 &none         &none
&none         &none         &none         &none         &none         &none         &none                  &none         &none         &bt BT_CLR    &bt BT_CLR    &bt BT_CLR    &none         &none
&none         &none         &none         &none         &none         &none         &none                  &none         &none         &none         &none         &none         &none         &none
                        >;
                };
        };
};

`

petejohanson, 2024-10-25:
You'll want to reference the behavior and update it, e.g.:

```
&lt {
    quick-tap-ms = <200>;
};

/ {
   the keymap goes here.
};
```

closed 2024-10-25 by petejohanson

flashlightbug, 2025-06-22:
Thank you, petejohanson, for posting this. This fixed my issue.