zmkfirmware/zmk #2618: CI: Add enforcement of conventional commit messages for PRs

petejohanson, 2024-11-08:
In order to effectively leverage release-please for release automation, we need to actually enforce the use of conventional commits for our PR/commits, so that release-please can do a proper job of generating changelogs, determining next release version number, etc.

Probably takes the form of an additional GH Action to check commit messages for PRs.

Nick-Munnich, 2024-11-08:
It might be nice to, on test failure, have a message informing/reminding new contributors about the details of conventional commits, along with listing helpful commands to rename and squash their commits accordingly.

caksoylar, 2024-11-08:
We should start by documenting this, maybe in a new page under "Development -> Contributing". https://github.com/zmkfirmware/zmk/blob/main/CONTRIBUTING.md already mentions it a bit, but I am not sure if people check there and I haven't been trying to update it either.

caksoylar, 2024-11-08:
As a side note, wouldn't automating checking certain commit types be a bit difficult? Like being able to tell what is a breaking change, when both types of files (say docs+boards) are touched in a PR, etc.

petejohanson, 2024-11-08:
> As a side note, wouldn't automating checking certain commit types be a bit difficult? Like being able to tell what is a breaking change, when both types of files (say docs+boards) are touched in a PR, etc.

I think at best this should be checking the *format*, not whether the selected type/area is correct. That will still require human verification, IHMO.

petejohanson, 2024-11-08:
Looking like gitlint is a nice option for this: https://jorisroovers.com/gitlint/latest/

Nick-Munnich, 2024-11-13:
Closed by #2623