zmkfirmware/zmk #2635: USB Powered MCU requires &bootloader to bootload

heynest, 2024-11-10:
Hi,

I tested the #2458 which works fine. So I have my keyboard currently powered by USB (nice!nano V2).
https://github.com/heynest/zmk-config-boardsource3x4/
It is handwired, see [discord for details](https://discord.com/channels/719497620560543766/1299665621591523393).

I found one limitation which should at least be listed here: https://zmk.dev/docs/keymaps/behaviors/outputs
If the keyboard is powered by USB (to my laptop of course) **I cannot get it into bootloader mode via tapping the reset button twice**. Luckily, my flashed layout had a &bootloader behavior in the initial layout, otherwise I would not be able to reflash it w/o major effort.

Freature request:
double reset should activate bootloader

Bug report:
https://zmk.dev/docs/keymaps/behaviors/outputs should mention the limitation (and what to do if one fall into the trap).

Thanks for your great work!

Nick-Munnich, 2024-11-10:
Hi, the requested feature is technically speaking not a part of ZMK. In addition, this behavior works fine for a majority of people, including myself. Make sure you are tapping it *twice* rapidly. It needs to be twice within 500ms. If you tap it an extra time, it may cause the board to reset itself and exit the bootloader. If this doesn't work, please reach out to whomever you bought the board from for assistance.

joelspadin, 2024-11-10:
More specifically, double tapping reset to enter bootloader is a function of the bootloader itself.

heynest, 2024-11-11:
Oh, dear. I just realized, that I connected my reset button to the wrong pins...
I retested and it works just fine.

Sorry for consuming your time unnecessarily!

petejohanson, 2024-11-11:
No worries. Closing.

closed 2024-11-11 by petejohanson