zmkfirmware/zmk #2652: Improve &bootloader Feature on Dongle-Connected Boards

kmandarin-orange, 2024-11-17:
I'm not knowledgeable enough about zmk to know if this is possible, so this might be a feature request, but I'd like to make the `&bootloader` feature work in the "expected" way with my [dongle-connected TOTEM split keyboard](https://github.com/eigatech/zmk-config/tree/totem-dongle). 

According to the ZMK docs, the bootloader feature only affects the side of the keyboard it's invoked from. This is useful on the standard Bluetooth split keeb that has a single "master" side, but is less useful when a dongle setup's master controller _is_ the dongle board itself, and that's the one you intend to re-flash with new keymaps. This made me wonder if there was some workaround to make this command target a specific micro controller, or if there would be a way to make a new function that'd trigger the bootloader on the master board specifically.

Nick-Munnich, 2024-11-17:
You can apply a workaround by binding bootloader to a combo, which always triggers the behavior on the central. Related: #2500

kmandarin-orange, 2024-11-17:
Something like this?

```c
Bootloader {
            bindings = <&bootloader>;
            key-positions = <20 34 25 35 31>;
        };
```

I've given this a shot now, but it doesn't seem to be working. I'm making sure to press all the keys at exactly the same time, but the dongle board isn't entering bootloader mode. Oddly though, all my other combos are working just fine. Am I making some kind of mistake here?

Nick-Munnich, 2024-11-17:
https://zmk.dev/docs/config/combos#kconfig Probably running into this.

kmandarin-orange, 2024-11-17:
Ah lord, should've double-checked the docs. After fixing that up, things worked as expected! Thank you so much for the help.

On the topic of documentation and the `&bootloader` feature, I stumbled on a related issue #1805 earlier, which talks some about changing some wording on the [&bootloader docs page](https://zmk.dev/docs/keymaps/behaviors/reset#bootloader-reset), and I agree it might be worth mentioning the details/logic of combo vs. `&bootloader` behavior. If that issue would be a better place to continue the discussion, I can close this issue and add to the conversation there. If that's out of the scope of that issue, I could rename this issue or make a new one to target a discussion about adding some information to that bootloader section.

caksoylar, 2024-11-17:
The information is already in two places: [combos page](https://zmk.dev/docs/keymaps/combos#advanced-usage) and [split keyboards feature page](https://zmk.dev/docs/features/split-keyboards#source-locality-behaviors). I guess it wouldn't hurt much to repeat it in https://zmk.dev/docs/keymaps/behaviors/reset#split-keyboards either, given this came up a few times.

kmandarin-orange, 2024-11-18:
Yeah, I missed the info in the combos page since I figured it'd be under the "Split Keyboards" section in the [Reset Behaviors page](https://zmk.dev/docs/keymaps/behaviors/reset#split-keyboards). 

Maybe to avoid duplicating wording, we could refer back to those pages from that section with something like "To always target the central board no matter where the shortcut is invoked from, try the [combo functionality](https://zmk.dev/docs/keymaps/combos#advanced-usage) and review [source locality documentation](https://zmk.dev/docs/features/split-keyboards#source-locality-behaviors)." Just having a short hook to make the connection to those other sections would probably help people connect those dots more easily.

caksoylar, 2024-11-18:
I think a mention of combos and a link could be good, agreed.

kmandarin-orange, 2024-11-19:
Sweet, that sounds good! I can go ahead and close this issue if you think it's served its purpose.

caksoylar, 2024-11-19:
Let's keep it open as a reminder to me (or anyone else) to add the pointer to the docs.

PR #2677 (2024-11-28): docs: Mention combos in reset behaviors
Fixes #2652.
commit 978c7cb1 (2024-11-28): docs: Mention combos in reset behaviors (#2677)

Fixes #2652
closed 2024-11-28 by Nick-Munnich