zmkfirmware/zmk #2653: Mod-tap key not working in mod-morph sequence

kmandarin-orange, 2024-11-17:
I'm having some trouble with getting any of my mod-morphs to work correctly, and it seems to be because of my shift key being behind a mod-tap. For example, I want to make SHIFT + BACKSPACE act as DELETE, but the behavior seems to simply be SHIFT and BACKSPACE being sent separately. That's what I gathered from this `wev` output at least:

```bash
[14:     wl_keyboard] key: serial: 91284; time: 50070095; key: 50; state: 1 (pressed)
                      sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 91285; group: 0
                      depressed: 00000001: Shift 
                      latched: 00000000
                      locked: 00000010: Mod2 
[14:     wl_keyboard] key: serial: 91286; time: 50070254; key: 22; state: 1 (pressed)
                      sym: BackSpace    (65288), utf8: '\b'
[14:     wl_keyboard] key: serial: 91287; time: 50070314; key: 22; state: 0 (released)
                      sym: BackSpace    (65288), utf8: ''
[14:     wl_keyboard] key: serial: 91288; time: 50070479; key: 50; state: 0 (released)
                      sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 91289; group: 0
                      depressed: 00000000
                      latched: 00000000
                      locked: 00000010: Mod2 
```

This behavior hasn't been limited to this specific case though, all of the following mod-morphs don't behave as expected when combined with either of my shift keys:

```c
behaviors {
        bksp_del: bksp_del {
            compatible = "zmk,behavior-mod-morph";
            label = "BKSP_DEL";
            bindings = <&kp BACKSPACE>, <&kp DELETE>;

            #binding-cells = <0>;
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };

        gaming_layer: gaming_layer {
            compatible = "zmk,behavior-mod-morph";
            label = "GAMING_LAYER";
            bindings = <&tog 2>, <&tog 3>;

            #binding-cells = <0>;
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };

        up_pgup: up_pgup {
            compatible = "zmk,behavior-mod-morph";
            label = "UP_PGUP";
            bindings = <&kp UP_ARROW>, <&kp PG_UP>;

            #binding-cells = <0>;
            mods = <(MOD_RSFT|MOD_LSFT)>;
        };

        down_pgdown: down_pgdown {
            compatible = "zmk,behavior-mod-morph";
            label = "DOWN_PGDOWN";
            bindings = <&kp DOWN_ARROW>, <&kp PAGE_DOWN>;

            #binding-cells = <0>;
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };

        left_home: left_home {
            compatible = "zmk,behavior-mod-morph";
            label = "LEFT_HOME";
            bindings = <&kp LEFT_ARROW>, <&kp HOME>;

            #binding-cells = <0>;
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };

[... 165 lines cut ...]
        movement_and_media {
            bindings = <
        &kp F1       &kp F2          &kp F3        &kp F4           &kp F5                  &kp F8                &kp F9          &kp UP_ARROW    &kp F11          &kp F12
        &kp F7       &kp K_MUTE      &kp C_VOL_DN  &kp C_VOLUME_UP  &kp C_BRIGHTNESS_DEC    &kp C_BRIGHTNESS_INC  &kp LEFT_ARROW  &kp DOWN_ARROW  &kp RIGHT_ARROW  &kp INSERT
&kp F6  &bootloader  &kp C_PREVIOUS  &kp C_PAUSE   &kp C_NEXT       &kp PRINTSCREEN         &trans                &trans          &trans          &trans           &trans      &kp K_LOCK
                                     &tog 3        &trans           &trans                  &trans                &trans          &trans
            >;
        };

        gaming {
            bindings = <
           &kp T  &kp Q  &kp W                &kp E      &kp R        &kp F1        &kp F2           &kp F3        &kp F4        &kp F5
           &kp G  &kp A  &kp S                &kp D      &kp F        &kp NUMBER_1  &kp NUMBER_2     &kp NUMBER_3  &kp NUMBER_4  &kp N5
&kp LSHFT  &kp B  &kp Z  &kp X                &kp C      &kp V        &kp H         &kp N            &kp M         &kp DOT       &kp FSLH  &kp RET
                         &mt LEFT_ALT ESCAPE  &kp LCTRL  &kp SPACE    &kp SPACE     &lt 2 BACKSPACE  &tog 0
            >;
        };

        // ▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄
        // ▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄
        // ▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄▄
    };
};
```
</details>

caksoylar, 2024-11-17:
Unless I am misreading, you are defining the mod-morph such as `bksp_del` but you aren't assigning it to the backspace key position. Mod morphs aren't global "rules", they are behaviors so you need to assign and use them for them to work.

kmandarin-orange, 2024-11-18:
Oh jeez that was a huge misunderstanding on my part, sorry to bother with a goofy issue report. I'll get this closed, thank you for your time!

closed 2024-11-18 by kmandarin-orange