zmkfirmware/zmk #2661: MacOS 15 Sequoia Bluetooth Latency 0

xudongzheng, 2024-11-19:
MacOS 15 Sequoia seems to be negotiating a Bluetooth connection latency of 0 instead of the usual 30. The primary impact is battery life.

Increasing `CONFIG_BT_CONN_PARAM_UPDATE_TIMEOUT` (for example to 12000ms) doesn't seem to make a difference. However I observed the latency value changing correctly in a different Zephyr application using `bt_conn_le_param_update()` manually after the connection is established. Perhaps ZMK needs to do that, assuming a future macOS update does not revert to the previous behavior.

Filing this issue until I or someone else has time to investigate more thoroughly.

felipesabino, 2025-03-17:
I am having a similar issue, it disconnects immediately after connecting

```
default	20:38:09.041333+0100	bluetoothd	macLeDeviceConnectionParametersUpdated: LE HID "Corne" current 30.00 ms latency(0) appearanceValue(0x3c1) updated
default	20:38:09.042253+0100	bluetoothd	Device ready - {cbuuid: 26B56745-1A82-3311-80AA-73A1C0CC8861, devicename: Corne, lmHandle: 0x43, adv-addr: EE:59:9C:3C:D8:05-Random, resolved-addr: , result: 0}
default	20:38:09.048500+0100	bluetoothd	Device found: CBDevice 26B56745-1A82-3311-80AA-73A1C0CC8861, BDA EE:59:9C:3C:D8:05, Nm 'Corne', PID 0x615E (?), VID 0x1D50, VS 1, DsFl 0x200000 < Connections >, DvF 0x40000210000 < HIDGoodBehavior Central Connectable >, CnS 0x400000 < BLE >, Aprc Keyboard, BTv 5.4, CF 0x80804200000 < Connections New DiscoveryFlags Attributes >
default	20:38:09.574388+0100	bluetoothd	DIS for device 26B56745-1A82-3311-80AA-73A1C0CC8861 DIS Model number: Corne
default	20:38:09.643166+0100	bluetoothd	isLowLatencyBLEHID: Device "Corne" is appearanceValue(0x3c1)
default	20:38:09.643393+0100	bluetoothd	isLowLatencyBLEHID: Device "Corne" is VID(0x1d50) / PID(0x615e)
default	20:38:09.643467+0100	bluetoothd	isLowLatencyBLEHID: Device "Corne" is appearanceValue(0x3c1)
default	20:38:09.643659+0100	bluetoothd	isLowLatencyBLEHID: Device "Corne" is VID(0x1d50) / PID(0x615e)
default	20:38:09.643794+0100	bluetoothd	switchLEHIDsConnectionInterval_block_invoke: Device "Corne" is Compatible LE HID
default	20:38:09.644040+0100	bluetoothd	switchLEHIDsConnectionInterval_block_invoke: LE HID "Corne" switching to 15.00 ms latency(0) appearanceValue(0x3c1) CELength(4)
default	20:38:09.697423+0100	BTLEServer	Updated DeviceInformation for Corne‘s ModelNumber - Corne
default	20:38:09.735582+0100	BluetoothUIServer	LEHIDConnectionComplete - VID:0x1D50 PID:0x615E “Corne”: no custom info found
default	20:38:09.755321+0100	BTLEServer	Updated DeviceInformation for Corne‘s ManufacturerName - ZMK Project
default	20:38:09.815795+0100	BTLEServer	Updated DeviceInformation for Corne‘s PnP - PID:0x615E VID:0x1D50
default	20:38:09.972258+0100	bluetoothd	macLeDeviceConnectionParametersUpdated: LE HID "Corne" cache device's preference CI is 15.00 ms
default	20:38:09.972311+0100	bluetoothd	isLowLatencyBLEHID: Device "Corne" is appearanceValue(0x3c1)
default	20:38:09.972402+0100	bluetoothd	isLowLatencyBLEHID: Device "Corne" is VID(0x1d50) / PID(0x615e)
default	20:38:09.972470+0100	bluetoothd	isLowLatencyBLEHID: Device "Corne" is appearanceValue(0x3c1)
default	20:38:09.972544+0100	bluetoothd	isLowLatencyBLEHID: Device "Corne" is VID(0x1d50) / PID(0x615e)
default	20:38:09.972589+0100	bluetoothd	macLeDeviceConnectionParametersUpdated: LE HID "Corne" current 15.00 ms latency(0) appearanceValue(0x3c1) updated
default	20:38:10.089116+0100	bluetoothd	Device disconnected - {cbuuid: 26B56745-1A82-3311-80AA-73A1C0CC8861, devicename: Corne, lmHandle: 0x43, adv-addr: EE:59:9C:3C:D8:05-Random, resolved-addr: , result: 0}
default	20:38:10.100383+0100	bluetoothd	Device lost: CBDevice 26B56745-1A82-3311-80AA-73A1C0CC8861, BDA EE:59:9C:3C:D8:05, Nm 'Corne', PID 0x615E (?), VID 0x1D50, VS 1, DvF 0x40000210000 < HIDGoodBehavior Central Connectable >, CnS 0x400000 < BLE >, Aprc Keyboard, BTv 5.4, CF 0x800200000 < Connections DiscoveryFlags >
```

I tried all flashing with all sorts of config variables, and I tried combining them differently as well according to some answers I found online, but nothing seems to make any difference

```
CONFIG_ZMK_BLE_CLEAR_BONDS_ON_START=y
CONFIG_BT_CTLR_TX_PWR_PLUS_8=y
CONFIG_BT_PERIPHERAL_PREF_MIN_INT=24
CONFIG_BT_PERIPHERAL_PREF_MAX_INT=36
CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
```

xudongzheng, 2025-03-17:
@felipesabino The impact of this issue is shorter battery life - it's not really related to being or not being able to connect. It sounds like you're experiencing a different issue. I recommend seeking help on the ZMK Discord.