zmkfirmware/zmk #2701: macOS specific keycodes does not work as expected

mr2rm, 2024-12-09:
I added the following keymaps on Kinesis Advantage 360 Pro to enable [Application Controls](https://zmk.dev/docs/keymaps/list-of-keycodes#application-controls) available on the MacBook keyboard and I used a [keymap editor](https://nickcoutsos.github.io/keymap-editor/) to do so. But, I couldn't see any events detected in Karabinar related to those keys, and they don't work as expected.

### Keymaps:
- `&kp C_AC_SEARCH`: Toggle Spotlight search - `{"key_code": "f4"}`
- `&kp C_AC_DESKTOP_SHOW_ALL_WINDOWS`: Show all the windows, - `{"key_code": "f3"}`
- `&kp C_AC_DESKTOP_SHOW_ALL_APPLICATIONS`:  Show all applications?
- `&kp C_AC_NEXT_KEYBOARD_LAYOUT_SELECT`: Select the next keyboard layout (Apple Globe) - `{"apple_vendoer_top_case_key_code": "keyboard_fn"}`

Also, here are my changes for creating a custom firmware that I used: https://github.com/mr2rm/Adv360-Pro-ZMK/commit/9cb9ddbd6ded8c64a0e815394ccf10ce20d38c7a

[image]

galaxyeden, 2025-05-11:
I haven't looked closely at this, but what's the USB PID and VID you have set? I believe that at least for fn/globe the key code is ignored unless it's an Apple keyboard.

What I do is use Karabiner to remap other combos. F20-F24 aren't supported by macOS but Karabiner does see them so they're great for this. Here's an example using cmd-shift-f23 (what Microsoft keyboards call the Copilot key), I have right command and left shift so my rule order doesn't matter (I also swap left command and left control in Karabiner):

```
{
    "description": "change copilot to fn",
    "manipulators": [
        {
            "conditions": [
                {
                    "identifiers": [{ "is_built_in_keyboard": true }],
                    "type": "device_unless"
                }
            ],
            "from": {
                "key_code": "f23",
                "modifiers": { "mandatory": ["right_command", "left_shift"] }
            },
            "to": [{ "key_code": "fn" }],
            "type": "basic"
        }
    ]
}
```

ReFil, 2025-05-14:
When using NKRO you need to enable extended reporting to get to some of the funkier consumer keys. Change this config line from =n to =y https://github.com/mr2rm/Adv360-Pro-ZMK/blob/60884c4f5f4b29eba96e5a802b9ea99e147d339d/config/boards/arm/adv360/adv360_left_defconfig#L65

Note you may also need to unpair and re-pair to the mac to refresh the descriptor