zmkfirmware/zmk #2712: How to control the level of GPIO in ZMK, during active and deep sleep.

PuzzleKid, 2024-12-14:
I have an LDO to power external devices, and I want to maintain power during deep sleep, so I want to be able to control a gpio to output high levels under any circumstances, does ZMK currently support this feature? If so, how to implement it?

PuzzleKid, 2024-12-20:
@
Dear Developer,
Could you please provide assistance with the issues? Thank you in advance for your help.

PuzzleKid, 2024-12-20:
@caksoylar @Nick-Munnich 
Dear Developer,
Could you please provide help with the issues? Thank you very much!

joelspadin, 2024-12-20:
You should be able to use a GPIO hog to force a pin to a specific state: https://docs.zephyrproject.org/latest/build/dts/api-usage.html#:~:text=node%2D1%20%7B%0A%3B%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20gpio%2Dhog%3B%0A%3B%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20gpios%20%3D%20%3C0x0%200x10%3E%2C%20%3C0x1%200x20%3E%3B%0A%3B%20%20%20%20%20%20%20%20%20%20%20%20%20%20%20output%2Dhigh%3B%0A%3B%20%20%20%20%20%20%20%7D%3B