zmkfirmware/zmk #2716: West Build error after latest commit: "fatal error: dt-bindings/zmk/pointing.h: No such file or directory"

ng-nicholas, 2024-12-16:
I had started to get the following error when committing my latest keymap changes for the Github action to run this week. For some reason, the build of the firmware for my left/right shield fails after discovering the device overlay step with the error highlighted in the title. The full log of the build is as follows:

> Run west build -s zmk/app -d "/tmp/tmp.jIKFsRLnoY" -b "sofle_right"  -- -DZMK_CONFIG=/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config -DSHIELD="nice_view_adapter nice_view_custom" 
> + west build -s zmk/app -d /tmp/tmp.jIKFsRLnoY -b sofle_right -- -DZMK_CONFIG=/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config -DSHIELD=nice_view_adapter nice_view_custom
> -- west build: generating a build system
> Loading Zephyr default modules (Zephyr base).
> -- Application: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zmk/app
> -- CMake version: 3.30.0
> -- Found Python3: /usr/bin/python3 (found suitable version "3.[1](https://github.com/ng-nicholas/sofle-j-keyboard-zmk/actions/runs/12345816418/job/34450491601#step:11:1)2.3", minimum required is "3.8") found components: Interpreter
> -- Cache files will be written to: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zephyr/.cache
> -- Zephyr version: 3.5.0 (/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zephyr)
> -- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
> -- Adding ZMK config directory as board root: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config
> -- Adding /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/urchin-peripheral-animation/boards/shields/nice_view_custom
> -- Adding /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zmk/app/boards/shields/nice_view_adapter
> -- Adding /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/boards/shields/nice_view_adapter
> -- ZMK Config directory: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config
> -- ZMK Config Kconfig: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/sofle.conf
> -- Using keymap file: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/sofle.keymap
> -- Board: sofle_right
> -- Shield(s): nice_view_adapter nice_view_custom
> -- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
> -- Found host-tools: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
> -- Found toolchain: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
> -- Found Dtc: /opt/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
> -- Found BOARD.dts: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/boards/arm/sofle/sofle_right.dts
> -- Found devicetree overlay: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/boards/shields/nice_view_adapter/nice_view_adapter.overlay
> -- Found devicetree overlay: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/boards/shields/nice_view_adapter/boards/sofle_right.overlay
> -- Found devicetree overlay: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/urchin-peripheral-animation/boards/shields/nice_view_custom/nice_view_custom.overlay
> -- Found devicetree overlay: /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/sofle.keymap
> In file included from <command-line>:
> /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/sofle.keymap:7:10: fatal error: dt-bindings/zmk/pointing.h: No such file or directory
>     7 | #include <dt-bindings/zmk/pointing.h>
>       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> CMake Error at /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zephyr/cmake/modules/extensions.cmake:3885 (message):
>   failed to preprocess devicetree files (error code 1):
>   /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/boards/arm/sofle/sofle_right.dts;/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/boards/shields/nice_view_adapter/nice_view_adapter.overlay;/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/boards/shields/nice_view_adapter/boards/sofle_right.overlay;/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/urchin-peripheral-animation/boards/shields/nice_view_custom/nice_view_custom.overlay;/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config/sofle.keymap
> Call Stack (most recent call first):
>   /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zephyr/cmake/modules/dts.cmake:226 (zephyr_dt_preprocess)
>   /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zephyr/cmake/modules/zephyr_default.cmake:[12](https://github.com/ng-nicholas/sofle-j-keyboard-zmk/actions/runs/12345816418/job/34450491601#step:11:13)9 (include)
>   /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
>   /__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
>   CMakeLists.txt:8 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/tmp.jIKFsRLnoY -GNinja -DBOARD=sofle_right -DZMK_CONFIG=/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/config '-DSHIELD=nice_view_adapter nice_view_custom' -S/__w/sofle-j-keyboard-zmk/sofle-j-keyboard-zmk/zmk/app
> Error: Process completed with exit code 1.

I had not made any significant changes to my keymap since the last successful build, so I was not expecting an error to occur. For reference, here are other details of my build:

- Board: Nice Nano v2
- Shield: Sofle with Nice View adapters
- Build repo (mine): https://github.com/ng-nicholas/sofle-j-keyboard-zmk
- Keymap editing done via Nick Coutsos' keymap editor: https://nickcoutsos.github.io/keymap-editor/

caksoylar, 2024-12-16:
You are pointing to an old mouse keys branch in https://github.com/ng-nicholas/sofle-j-keyboard-zmk/blob/main/config/west.yml#L10. It looks like keymap editor assumes you use ZMK `main` and automatically added that header include in https://github.com/ng-nicholas/sofle-j-keyboard-zmk/commit/a8f2bdfa7be4d3691c92afce983b61723d500404#diff-51c9bbf1231f078c044540d5755ef9abb0734a6303336344ff63e0f187f7ab3eR7.

Now that mouse keys are merged, you can [switch to using ZMK main](https://github.com/zmkfirmware/unified-zmk-config-template/blob/main/config/west.yml) and the issue should disappear.

ng-nicholas, 2024-12-16:
@caksoylar Got it, thank you for the advice!

closed 2024-12-16 by ng-nicholas