zmkfirmware/zmk #2719: zip_temp_layer sometimes stop working

jhorology, 2024-12-17:
zip_temp_layer sometimes stops working. I think the cause is that input_processor_temp_layer.c is not thread-safe. CONFIG_INPUT_MODE_THREAD is Zephyr's default, but is the pointing subsystem supposed to use CONFIG_INPUT_MODE_SYNCHRONOUS and the input driver to use the system thread?

my configuration:
https://github.com/jhorology/keyboards/blob/main/zmk_keyboards/boards/shields/libra_mini/libra_mini.overlay
https://github.com/jhorology/keyboards/blob/main/zmk_keyboards/libra_mini.keymap

petejohanson, 2024-12-20:
@jhorology Can you please give #2729 a try?

AYM1607, 2024-12-23:
Just in case it serves as another data point, I think that this comment sheds some light on the actual problem: https://github.com/zmkfirmware/zmk/pull/2729#issuecomment-2558130061 for me a consistent repro is invoking a `&to some-layer` behavior while the temp layer is active and _has not  off due to the timer_, this causes the temp layer to never be enabled again and requires a keyboard reset.

RealET, 2025-01-11:
I have also reproduced the `&to some-layer` issue @AYM1607.

AYM1607, 2025-01-12:
If you haven't seen this already @RealET, I linked a couple commits on https://github.com/zmkfirmware/zmk/pull/2729 that fix the issue for me. If you want to temporarily circumvent it, here's how I'm building my firmware based on the patch I mentioned above: https://github.com/HumanComputingInc/kb42s_default_config/blob/main/config/west.yml

PR #2729 (2025-03-01): fix(pointing): Temp layer threading protection.
* Ensure all layer operations occur on the system work queue thread.

Fixes: #2719

## PR check-list

- [x] Branch has a [clean commit history](https://zmk.dev/docs/development/contributing/pull-requests#clean-commit-history)
- [x] Additional tests are included, if changing behaviors/core code that is testable.
- [x] Proper Copyright + License headers added to applicable files (Generally, we stick to "The ZMK Contributors" for copyrights to help avoid churn when files get edited)
- [x] [Pre-commit](https://zmk.dev/docs/development/local-toolchain [... cut]
commit 1e3e62c1 (2025-03-01): fix(pointing): Temp layer threading protection. (#2729)

fix(pointing): Temp layer threading protection.

Ensure all layer operations occur on the system work queue thread.

Fixes: #2719

fix(pointing): Handle layer events to disable events

Make the temp layer input processor propely handle layers getting
deactivated externally before the temp layer timeout.
closed 2025-03-01 by petejohanson