zmkfirmware/zmk #2735: Hold-Tap with &caps_word - lacks binding in keymap

elXaptation, 2024-12-27:
Hi, not sure what I'm doing wrong here of if something is broken. 

I'm trying to get my SHIFT key to be sticky on tap and go Caps_Word if I hold it. 

Seems I'm using Hold-Tap with parameter-less behaviors correctly according to [the docs](https://zmk.dev/docs/keymaps/behaviors/hold-tap#using-different-behavior-types-with-hold-taps). "Dummy Parameter"

Github Actions build fails with "lacks binding" in node &caps when referenced by node keymap:Base.

Seems a similar situation was working in 2024-Oct, [Issue 2577]( https://github.com/zmkfirmware/zmk/issues/2577).

```
//Make Sticky Keys lazy. 
&sk {
    release-after-ms = <2000>;
    quick-release;
    lazy;
    /delete-property/ ignore-modifiers;
};

/ {
    behaviors {
        // Sticky SHFT with CapsWord
        caps: capsword_with_sticky_shift {
                compatible = "zmk,behaviour-hold-tap";
                #binding-cells = <2>;
                flavor = "tap-preferred";
                tapping-term-ms = <150>;
                quick-tap-ms = <0>;
                bindings = <&caps_word>,<&sk>;
        };
    };
};
/ {
        keymap {
                compatible = "zmk,keymap";
                Base {
                        display-name = "QWERTY Base";
                        bindings = < ... &caps 0 LSHFT ... >;
                };
       };
};
```

Build Logs:
``` devicetree error: binding controller <Node /behaviors/capsword_with_sticky_shift in '/__w/CorneKeyboard/CorneKeyboard/zephyr/misc/empty_file.c'> for <Node /keymap/Base in '/__w/CorneKeyboard/CorneKeyboard/zephyr/misc/empty_file.c'> lacks binding
CMake Error at /__w/CorneKeyboard/CorneKeyboard/zephyr/cmake/modules/dts.cmake:279 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /__w/CorneKeyboard/CorneKeyboard/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /__w/CorneKeyboard/CorneKeyboard/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /__w/CorneKeyboard/CorneKeyboard/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:9 (find_package)

-- In: /tmp/tmp.4vncWImdEM/zephyr, command: /usr/bin/python3;/__w/CorneKeyboard/CorneKeyboard/zephyr/scripts/dts/gen_defines.py;--dts;/tmp/tmp.4vncWImdEM/zephyr/zephyr.dts.pre;--dtc-flags;'-Wno-unique_unit_address_if_enabled';--bindings-dirs;/__w/CorneKeyboard/CorneKeyboard/zmk/app/module/dts/bindings;/__w/CorneKeyboard/CorneKeyboard/zmk/app/dts/bindings;/__w/CorneKeyboard/CorneKeyboard/zephyr/dts/bindings;--header-out;/tmp/tmp.4vncWImdEM/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/tmp/tmp.4vncWImdEM/zephyr/zephyr.dts.new;--edt-pickle-out;/tmp/tmp.4vncWImdEM/zephyr/edt.pickle;--vendor-prefixes;/__w/CorneKeyboard/CorneKeyboard/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/__w/CorneKeyboard/CorneKeyboard/zephyr/dts/bindings/vendor-prefixes.txt
-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/tmp.4vncWImdEM -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/__w/CorneKeyboard/CorneKeyboard/config '-DSHIELD=corne_left nice_view_adapter nice_view' -DSNIPPET=studio-rpc-usb-uart -S/__w/CorneKeyboard/CorneKeyboard/zmk/app
Error: Process completed with exit code 1.```

caksoylar, 2024-12-27:
Going by just the config snippet here (since the error message you posted isn't the informative part) your `compatible` value uses "behaviour" with "u", not the spelling used by ZMK "behavior".

elXaptation, 2024-12-27:
Ah yes.. thank you. Of course it would be something like this. I must have
stared at this typo for hours. This is why I need a better keyboard....

I appreciate the response. Will confirm my error shortly..

elXaptation, 2024-12-27:
Confirmed.

Fixing the typo resolved the build failure.

This one is ready to close as PEBKAC. Thank again.

closed 2024-12-27 by Nick-Munnich