zmkfirmware/zmk #2746: Make it more beginnerfriendly

Donnervogel, 2025-01-03:
I would love to see some cheap and beginnerfriendly and cheap boards supportet. Like the pi pico w, waveshare rp2040 zero, waveshare esp32-c6-mini, DFRobot Beetle ESP 32 C6. Because not everyone has the money to buy an nice nano or something simular. So these Boards would make it easier to beginn with custom keyboards and zmk. Espacially for students. Sadly i cant really Code, so i cant implement the boards myself.

Nick-Munnich, 2025-01-03:
For boards, we typically rely on Zephyr. You can see a list of boards that Zephyr supports for our current version of Zephyr [here](https://docs.zephyrproject.org/3.5.0/boards/index.html#boards), and you can see the list once we update to the most recent version [here](https://docs.zephyrproject.org/latest/boards/index.html#boards=). The pi pico (non-w), seeeduino series, and some blackpills are examples of cheap and beginner friendly boards that our current version of Zephyr (and thus ZMK) supports.

ilikeheaps, 2025-01-03:
Also there are cheaper boards similar to nice nano: https://github.com/joric/nrfmicro/wiki/Alternatives#supermini-nrf52840

caksoylar, 2025-01-03:
Regarding specifically ESP32 chips, they have limitations that make it impractical for battery-powered (or not) keyboard use (from @xudongzheng on Discord):

- Power consumption for all three are pretty bad, maybe 100x-1000x compared to Nordic microcontrollers, so not great for battery-powered builds
- ESP32-C3 and ESP32-C6 lack USB
- ESP32-S3 technically supports USB, but not with Zephyr ATM

myst729, 2025-01-13:
> Regarding specifically ESP32 chips, they have limitations that make it impractical for battery-powered (or not) keyboard use (from @xudongzheng on Discord):
>
> * Power consumption for all three are [...]

I guess ESP chips are not designed typically for wearable devices. They are more feasible for devices that have steady DC power supply (data over the air but power through wire), e.g. smart controls in your house or desktop.

mhantsch, 2025-02-06:
> cheaper boards similar to nice nano: https://github.com/joric/nrfmicro/wiki/Alternatives#supermini-nrf52840

Yes, the Supermini work really nicely. Pro micro compatible footprint, and behave as nice_nano_v2 in ZMK.

peterjc, 2025-08-02:
For the Raspberry Pi Pico (one of the larger RP2040 controllers), see ZMK tracking issue #1085