zmkfirmware/zmk #2751: header files not found in compilation

TimoWielink, 2025-01-05:
I am getting this error when trying to compile:

```
Run west build -s zmk/app -d "/tmp/tmp.ZjKc84dMKX" -b "nice_nano_v2"  -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD="minimalpad" -DZMK_EXTRA_MODULES='/__w/zmk-config-minimalpad/zmk-config-minimalpad' 
+ west build -s zmk/app -d /tmp/tmp.ZjKc84dMKX -b nice_nano_v2 -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD=minimalpad -DZMK_EXTRA_MODULES=/__w/zmk-config-minimalpad/zmk-config-minimalpad
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /tmp/zmk-config/zmk/app
-- CMake version: 3.30.0
-- Found Python3: /usr/bin/python3 (found suitable version "3.[12](https://github.com/TimoWielink/zmk-config-minimalpad/actions/runs/12621898752/job/35169231827#step:11:13).3", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /tmp/zmk-config/zephyr/.cache
-- Zephyr version: 3.5.0 (/tmp/zmk-config/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.[14](https://github.com/TimoWielink/zmk-config-minimalpad/actions/runs/12621898752/job/35169231827#step:11:15).0")
-- Adding /__w/zmk-config-minimalpad/zmk-config-minimalpad/boards/shields/minimalpad
-- ZMK Config directory: /tmp/zmk-config/config
-- Using keymap file: /__w/zmk-config-minimalpad/zmk-config-minimalpad/boards/shields/minimalpad/minimalpad.keymap
-- Board: nice_nano_v2
-- Shield(s): minimalpad
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.[16](https://github.com/TimoWielink/zmk-config-minimalpad/actions/runs/12621898752/job/35169231827#step:11:17).3 (/opt/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found Dtc: /opt/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /tmp/zmk-config/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /__w/zmk-config-minimalpad/zmk-config-minimalpad/boards/shields/minimalpad/minimalpad.overlay
-- Found devicetree overlay: /__w/zmk-config-minimalpad/zmk-config-minimalpad/boards/shields/minimalpad/minimalpad.keymap
In file included from <command-line>:
/__w/zmk-config-minimalpad/zmk-config-minimalpad/boards/shields/minimalpad/minimalpad.overlay:1:10: fatal error: dt-bindings/zmk/kscan.h: No such file or directory
    1 | #include <dt-bindings/zmk/kscan.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
-- Configuring incomplete, errors occurred!
CMake Error at /tmp/zmk-config/zephyr/cmake/modules/extensions.cmake:3885 (message):
  failed to preprocess devicetree files (error code 1):
  /tmp/zmk-config/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts;/__w/zmk-config-minimalpad/zmk-config-minimalpad/boards/shields/minimalpad/minimalpad.overlay;/__w/zmk-config-minimalpad/zmk-config-minimalpad/boards/shields/minimalpad/minimalpad.keymap
Call Stack (most recent call first):
  /tmp/zmk-config/zephyr/cmake/modules/dts.cmake:[22](https://github.com/TimoWielink/zmk-config-minimalpad/actions/runs/12621898752/job/35169231827#step:11:23)6 (zephyr_dt_preprocess)
  /tmp/zmk-config/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:9 (find_package)

FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/tmp.ZjKc84dMKX -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD=minimalpad -DZMK_EXTRA_MODULES=/__w/zmk-config-minimalpad/zmk-config-minimalpad -S/tmp/zmk-config/zmk/app
Error: Process completed with exit code 1.
``` 

I haveused the starter template and this is my .overlay file

```
#include <dt-bindings/zmk/kscan.h>
#include <dt-bindings/zmk/matrix_transform.h>

/ {
    kscan0: kscan0 {
        compatible = "zmk,kscan-gpio-matrix";
        label = "KSCAN";
        diode-direction = "col2row"; // Adjust if necessary based on diode direction
        row-gpios = <
            &pro_micro 20 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* ROW0 */
            &pro_micro 19 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* ROW1 */
            &pro_micro 18 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* ROW2 */
            &pro_micro 17 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* ROW3 */
        >;
        col-gpios = <
            &pro_micro 8 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)  /* COL0 */
            &pro_micro 9 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)  /* COL1 */
            &pro_micro 10 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* COL2 */
            &pro_micro 11 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* COL3 */
        >;
    };

    keymap_transform: keymap_transform {
        compatible = "zmk,matrix-transform";
        label = "Keymap Transform";
        rows = <4>;
        columns = <4>;
        map = <
            RC(0, 0) RC(0, 1) RC(0, 2) RC(0, 3) /* Row 0 */
            RC(1, 0) RC(1, 1) RC(1, 2) RC(1, 3) /* Row 1 */
            RC(2, 0) RC(2, 1) RC(2, 2) RC(2, 3) /* Row 2 */
            RC(3, 0) RC(3, 1) RC(3, 2) RC(3, 3) /* Row 3 */
        >;
    };

    leds {
        compatible = "zmk,ws2812";
        label = "Key LEDs";
        chain-length = <16>;  /* Number of SK6812 LEDs */
        io-pin = <&pro_micro 5 GPIO_ACTIVE_HIGH>; /* KEY_LEDS pin */
    };

    encoder0: encoder0 {
        compatible = "zmk,rotary-encoder";
        label = "Encoder0";
        a-pin = <&pro_micro 7 GPIO_ACTIVE_HIGH>; /* ENCODER0_L1 */
        b-pin = <&pro_micro 14 GPIO_ACTIVE_HIGH>; /* ENCODER0_L0 */
    };

    chosen {
        zmk,kscan = &kscan0;
        zmk,physical-layout = &keymap_transform;
    };
};

```

caksoylar, 2025-01-05:
Which starter template did you use? `#include <dt-bindings/zmk/kscan.h>` is not a header file and I can't find it documented anywhere.

TimoWielink, 2025-01-06:
Its inside of my .overlay file

```
#include <dt-bindings/zmk/matrix_transform.h>
#include <dt-bindings/zmk/kscan.h>

/ {
    kscan0: kscan0 {
        compatible = "zmk,kscan-gpio-matrix";
        label = "KSCAN";
        diode-direction = "col2row"; // Adjust if necessary based on diode direction
        row-gpios = <
            &pro_micro 20 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* ROW0 */
            &pro_micro 19 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* ROW1 */
            &pro_micro 18 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP) /* ROW2 */

```

TimoWielink, 2025-01-06:
@caksoylar As in this file https://github.com/caksoylar/zmk-config/blob/main/boards/shields/grumpybird/grumpybird.overlay

Nick-Munnich, 2025-01-06:
> Its inside of my .overlay file

Yes, but *why* is it there? What source told you to put it there? As far as I'm aware that header does not exist, and that line should not be there. If there is a template or source telling you to do so, we would like to know about it.