zmkfirmware/zmk #2755: Feature request: Double-Tap and Hold, for mod-tap and layer-tap

sleevesluu, 2025-01-07:
In VIA, for mod-tap and layer-tap, if I double-tap and hold, it holds down the key command. For example: with MT(MOD_RCTL, KC_RGHT), if I were to double-tap and hold that key, it would hold down KC_RGHT.

KLingO-MS, 2025-01-07:
it exist in ZMK too but probably default timing is too tight within which You have to double tap and hold ... but You can create Your own behavior with whatever timing suits You (mod-tap and layer-tap are just specific types of hold-tap)
see: https://zmk.dev/docs/keymaps/behaviors/hold-tap

Nick-Munnich, 2025-01-07:
https://zmk.dev/docs/keymaps/behaviors/hold-tap#quick-tap-ms

sleevesluu, 2025-01-08:
I guess this was more of an issue for ZMK Studio. I'll add a feature request there to perhaps change the default timings to something more usable.

closed 2025-01-08 by sleevesluu