zmkfirmware/zmk #2777: Dongle configuration fails with DEVICE_NAME_GET(DT_N_INST_0_zmk_behavior_momentary_layer_FULL_NAME) too long

jmlvargas, 2025-01-19:
Hi guys,

I have been trying to create a dongle configuration for Corne so followed the Keyboard Dongle guide, but continue getting this build error:

`In file included from /tmp/zmk-config/zephyr/include/zephyr/toolchain.h:50,
                 from /tmp/zmk-config/zephyr/include/zephyr/sys/util.h:18,
                 from /tmp/zmk-config/zephyr/include/zephyr/devicetree.h:25,
                 from /tmp/zmk-config/zephyr/include/zephyr/device.h:12,
                 from /tmp/zmk-config/zmk/app/src/behaviors/behavior_momentary_layer.c:9:
/tmp/zmk-config/zephyr/include/zephyr/toolchain/gcc.h:87:36: error: static assertion failed: "DEVICE_NAME_GET(DT_N_INST_0_zmk_behavior_momentary_layer_FULL_NAME) too long"
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert(EXPR, "" MSG)
      |                                    ^~~~~~~~~~~~~~
/tmp/zmk-config/zephyr/include/zephyr/device.h:877:9: note: in expansion of macro 'BUILD_ASSERT'
  877 |         BUILD_ASSERT(sizeof(Z_STRINGIFY(name)) <= Z_DEVICE_MAX_NAME_LEN,       \
      |         ^~~~~~~~~~~~
/tmp/zmk-config/zephyr/include/zephyr/device.h:998:9: note: in expansion of macro 'Z_DEVICE_NAME_CHECK'
  998 |         Z_DEVICE_NAME_CHECK(name);                                             \
      |         ^~~~~~~~~~~~~~~~~~~
/tmp/zmk-config/zephyr/include/zephyr/device.h:181:9: note: in expansion of macro 'Z_DEVICE_DEFINE'
  181 |         Z_DEVICE_DEFINE(node_id, Z_DEVICE_DT_DEV_ID(node_id),                  \
      |         ^~~~~~~~~~~~~~~
/tmp/zmk-config/zephyr/include/zephyr/device.h:196:9: note: in expansion of macro 'DEVICE_DT_DEFINE'
  196 |         DEVICE_DT_DEFINE(DT_DRV_INST(inst), __VA_ARGS__)
      |         ^~~~~~~~~~~~~~~~
/tmp/zmk-config/zmk/app/include/drivers/behavior.h:181:5: note: in expansion of macro 'DEVICE_DT_INST_DEFINE'
  181 |     DEVICE_DT_INST_DEFINE(inst, __VA_ARGS__);                                                      \
      |     ^~~~~~~~~~~~~~~~~~~~~
/tmp/zmk-config/zmk/app/src/behaviors/behavior_momentary_layer.c:68:1: note: in expansion of macro 'BEHAVIOR_DT_INST_DEFINE'
   68 | BEHAVIOR_DT_INST_DEFINE(0, behavior_mo_init, NULL, &behavior_mo_data, &behavior_mo_config,
      | ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.`

I followed the Dongle with Studio and also the Dongle without Studio instructions, in both cases got the same error. The keymap is the default one, no modifications, no custom stuff other than the result of the guide (as I understood it). 

This is my configuration repo: https://github.com/jmlvargas/zmk-config/tree/corne-dongle

Please let me know if I can help, although I'm not a C/C++ programmer.

Thank you

ReFil, 2025-01-19:
You'll get that issue if the build process isn't finding your keymap file, double check your config is structured correcly

caksoylar, 2025-01-19:
Indeed, if you want this keymap https://github.com/jmlvargas/zmk-config/blob/corne-dongle/boards/shields/jxcorne/corne.keymap to be used in the build the folder needs to be named `corne`, not `jxcorne`.

jmlvargas, 2025-01-19:
It is compiling now after applying the suggested changes.

Thank you all for the rapid answers!

closed 2025-01-19 by jmlvargas

AnyLameName, 2025-05-23:
@ReFil - Thank you! I am not sure I ever would have realized that a typo in my keymap filename was leading to THIS error, of all things.