zmkfirmware/zmk #2798: Mouse clicks (LCLK etc) don't work

siadat, 2025-01-26:
Mouse clicking used to work a few weeks ago, but it stopped working. I use Glove80's layout editor with the following keymap (search for LCLK, MCLK, RCLK)

```

/*
 * Copyright (c) 2020 The ZMK Contributors
 * Copyright (c) 2023 Innaworks Development Limited, trading as MoErgo
 *
 * SPDX-License-Identifier: MIT
 */

/* THIS FILE WAS GENERATED BY GLOVE80 LAYOUT EDITOR
 *
 * This file was generated automatically. You may or may not want to
 * edit it directly.
 */

#include <behaviors.dtsi>
/* Include all behavior includes needed */
#include <dt-bindings/zmk/outputs.h>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/rgb.h>

/* On demand includes */

/* Automatically generated layer name #define */
#define LAYER_QWERTY 0
#define LAYER_symboles 1
#define LAYER_Navigation 2
#define LAYER_Magic 3

/* To deal with the situation where there is no Lower layer, to keep &lower happy */
#ifndef LAYER_Lower
#define LAYER_Lower 0
#endif

/* Custom Device-tree */
#include <dt-bindings/zmk/mouse.h>

/* Glove80 system behavior & macros */
/ {
    behaviors {
        // For the "layer" key, it'd nice to be able to use it as either a shift or a toggle.
        // Configure it as a tap dance, so the first tap (or hold) is a &mo and the second tap is a &to
        lower: lower {
            compatible = "zmk,behavior-tap-dance";
            label = "LAYER_TAP_DANCE";
            #binding-cells = <0>;
            tapping-term-ms = <200>;
            bindings = <&mo LAYER_Lower>, <&to LAYER_Lower>;
        };
    };
};

/ {
    macros {
        rgb_ug_status_macro: rgb_ug_status_macro {
            label = "RGB_UG_STATUS";
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings
                = <&rgb_ug RGB_STATUS>;
        };
    };
};

/ {
#ifdef BT_DISC_CMD
    behaviors {
[... 244 lines cut ...]
        layer_Navigation {
            bindings = <
   &none   &none   &none   &none   &none                                                                           &kp C_MUTE     &kp C_VOL_DN  &kp C_VOL_UP              &none   &none
  &trans   &none   &none   &none   &none   &none                                                     &none              &none     &kp C_BRI_DN  &kp C_BRI_UP              &none   &none
 &kp ESC   &none  &kp N1  &kp N2  &kp N3   &none                                                     &none  &kp LS(LC(PG_UP))  &kp LS(LC(TAB))   &kp LC(TAB)  &kp LS(LC(PG_DN))   &none
  &trans  &kp N0  &kp N4  &kp N5  &kp N6   &none                                                  &kp LEFT           &kp DOWN           &kp UP     &kp RIGHT              &none  &trans
  &trans   &none  &kp N7  &kp N8  &kp N9   &none  &trans  &trans  &trans  &trans  &trans  &trans     &none              &none            &none         &none              &none  &trans
  &trans   &none   &none   &none   &none          &trans  &trans  &trans  &trans  &trans  &trans                        &none            &none         &none              &none  &trans
            >;
        };

        layer_Magic {
            bindings = <
  &bt BT_CLR            &none            &none            &none            &none                                                                                  &none   &none   &none   &none  &bt BT_CLR_ALL
       &none            &none            &none            &none            &none            &none                                                         &none   &none   &none   &none   &none           &none
       &none  &rgb_ug RGB_SPI  &rgb_ug RGB_SAI  &rgb_ug RGB_HUI  &rgb_ug RGB_BRI  &rgb_ug RGB_TOG                                                         &none   &none   &none   &none   &none           &none
 &bootloader  &rgb_ug RGB_SPD  &rgb_ug RGB_SAD  &rgb_ug RGB_HUD  &rgb_ug RGB_BRD  &rgb_ug RGB_EFF                                                         &none   &none   &none   &none   &none     &bootloader
  &sys_reset            &none            &none            &none            &none            &none   &bt_2   &bt_3         &none   &none   &none   &none   &none   &none   &none   &none   &none      &sys_reset
       &none            &none            &none            &none            &none                    &bt_0   &bt_1  &out OUT_USB   &none   &none   &none           &none   &none   &none   &none           &none
            >;
        };

    };
};
```

petejohanson, 2025-01-26:
I have no idea the state of the MoErgo fork, I'd recommend trying against ZMK `main` and if that works as expected, ask the MoErgo folks for help.

caksoylar, 2025-01-26:
I think it might be that the HID report descriptor was updated between their previous firmware versions and 24.01 that they released recently. Given that, it is worth trying https://zmk.dev/docs/features/bluetooth#refreshing-the-hid-descriptor. Also note that the header include has changed: https://zmk.dev/docs/keymaps/behaviors/mouse-emulation#mouse-emulation-defines

That being said, Pete is right, this is not the right venue to debug this. If you want to follow up please post on their [Discord](https://www.moergo.com/discord), I'll see it and respond.

siadat, 2025-01-26:
Thank you for the inputs and apologies for opening the issue here.

Just fyi: I think the moergo-sc/zmk fork is rebased each time, so unfortunately the history is rewritten and it is hard to find the previous working version of their fork. But I found a commit hash in a previous Actions run :) The latest one I found was this one https://github.com/moergo-sc/zmk/commit/fd3aa4b64f7cb1d5879a18dd0dda0330ef9f618f ([history](https://github.com/moergo-sc/zmk/commits/fd3aa4b64f7cb1d5879a18dd0dda0330ef9f618f)) in which the mouse clicks worked as built [running it like this](https://github.com/siadat/glove80-zmk-config/commit/33e13cde635f9b917c186e47df3d7f2be7446562#diff-4d2a8eefdf2a9783512a35da4dc7676a66404b6f3826a8af9aad038722da6823R11).

closed 2025-01-26 by siadat