zmkfirmware/zmk #2800: build-user-config failing: ModuleNotFoundError: No module named 'distutils'

jimrhoskins, 2025-01-28:
I have been building my zmk firmware via the github actions flow recommended by the documentation. It had been working well, but I went to make a change after several months and the build is failing. It seems like python-distutils is not found in the CI environment. I tried the same config that worked previously, but they all fail in the same way. 

[Example Failure](https://github.com/jimrhoskins/zmk-config/actions/runs/13021471303/job/36322849955)

Output:
```
Run west build -s zmk/app -d "/tmp/tmp.BG25H5oV21" -b "nice_nano_v2" -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD="corne_right" -DZMK_EXTRA_MODULES='/__w/zmk-config/zmk-config' 
+ west build -s zmk/app -d /tmp/tmp.BG25H5oV21 -b nice_nano_v2 -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD=corne_right -DZMK_EXTRA_MODULES=/__w/zmk-config/zmk-config
-- west build: generating a build system
-- ZMK Config directory: /tmp/zmk-config/config
-- Board: nice_nano_v2, /tmp/zmk-config/zmk/app/boards/arm/nice_nano, corne_right, corne
-- ZMK Config Kconfig: /tmp/zmk-config/config/corne.conf
-- Using keymap file: /tmp/zmk-config/config/corne.keymap
Including boilerplate (Zephyr base): /tmp/zmk-config/zephyr/cmake/app/boilerplate.cmake
CMake Deprecation Warning at /tmp/zmk-config/zephyr/cmake/app/boilerplate.cmake:37 (cmake_policy):
  The OLD behavior for policy CMP0079 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:17 (find_package)

-- Using keymap file: /tmp/zmk-config/config/corne.keymap
-- Application: /tmp/zmk-config/zmk/app
-- Zephyr version: 2.5.0 (/tmp/zmk-config/zephyr)
-- Found Python3: /usr/bin/python3.12 (found suitable exact version "3.12.3") found components: Interpreter
-- Found west (found suitable version "1.2.0", minimum required is "0.7.1")
-- git describe failed: fatal: No tags can describe 'f2108f1646b964adffb9c7ed74641c828fef2a44'.
Try --always, or create some tags.;
   BUILD_VERSION is left undefined
-- Board: nice_nano_v2, Shield(s): corne_right
-- Cache files will be written to: /tmp/zmk-config/zephyr/.cache
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found dtc: /opt/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /tmp/zmk-config/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /tmp/zmk-config/zmk/app/boards/shields/corne/corne_right.overlay
-- Found devicetree overlay: /tmp/zmk-config/config/corne.keymap
-- Generated zephyr.dts: /tmp/tmp.BG25H5oV21/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /tmp/tmp.BG25H5oV21/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /tmp/tmp.BG25H5oV21/zephyr/include/generated/device_extern.h
Parsing /tmp/zmk-config/zmk/app/Kconfig
Loaded configuration '/tmp/zmk-config/zmk/app/boards/arm/nice_nano/nice_nano_v2_defconfig'
Merged configuration '/tmp/zmk-config/config/corne.conf'
Merged configuration '/tmp/zmk-config/zmk/app/boards/shields/corne/corne_right.conf'
Configuration saved to '/tmp/tmp.BG25H5oV21/zephyr/.config'
Kconfig header saved to '/tmp/tmp.BG25H5oV21/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Deprecation Warning at /tmp/zmk-config/modules/lib/civetweb/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /tmp/zmk-config/modules/lib/civetweb/CMakeLists.txt:4 (cmake_policy):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done (3.3s)
CMake Warning (dev) at /tmp/zmk-config/zephyr/cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0[11](https://github.com/jimrhoskins/zmk-config/actions/runs/13021471303/job/36322849955#step:11:12)6 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  /tmp/zmk-config/zephyr/CMakeLists.txt:689 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /tmp/zmk-config/zephyr/cmake/linker/ld/target.cmake:33 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  /tmp/zmk-config/zephyr/CMakeLists.txt:1028 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done (0.1s)
CMake Warning:
  Manually-specified variables were not used by the project:

    ZMK_EXTRA_MODULES

-- Build files have been written to: /tmp/tmp.BG25H5oV21
-- west build: building application
[1/265] Preparing syscall dependency handling

[2/265] Generating misc/generated/syscalls_subdirs.trigger
[3/265] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[4/265] Generating include/generated/driver-validation.h
FAILED: zephyr/include/generated/driver-validation.h /tmp/tmp.BG25H5oV21/zephyr/include/generated/driver-validation.h 
cd /tmp/tmp.BG25H5oV21/zephyr && /usr/bin/python3.[12](https://github.com/jimrhoskins/zmk-config/actions/runs/13021471303/job/36322849955#step:11:13) /tmp/zmk-config/zephyr/scripts/gen_kobject_list.py --validation-output /tmp/tmp.BG25H5oV21/zephyr/include/generated/driver-validation.h --include /tmp/tmp.BG25H5oV21/zephyr/misc/generated/struct_tags.json
Traceback (most recent call last):
  File "/tmp/zmk-config/zephyr/scripts/gen_kobject_list.py", line 60, in <module>
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named 'distutils'
[5/265] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
FAILED: zephyr/include/generated/kobj-types-enum.h zephyr/include/generated/otype-to-str.h /tmp/tmp.BG25H5oV[21](https://github.com/jimrhoskins/zmk-config/actions/runs/13021471303/job/36322849955#step:11:22)/zephyr/include/generated/kobj-types-enum.h /tmp/tmp.BG25H5oV21/zephyr/include/generated/otype-to-str.h 
cd /tmp/tmp.BG25H5oV21/zephyr && /usr/bin/python3.12 /tmp/zmk-config/zephyr/scripts/gen_kobject_list.py --kobj-types-output /tmp/tmp.BG[25](https://github.com/jimrhoskins/zmk-config/actions/runs/13021471303/job/36322849955#step:11:26)H5oV21/zephyr/include/generated/kobj-types-enum.h --kobj-otype-output /tmp/tmp.BG25H5oV21/zephyr/include/generated/otype-to-str.h --kobj-size-output /tmp/tmp.BG25H5oV21/zephyr/include/generated/otype-to-size.h --include /tmp/tmp.BG25H5oV21/zephyr/misc/generated/struct_tags.json
Traceback (most recent call last):
  File "/tmp/zmk-config/zephyr/scripts/gen_kobject_list.py", line 60, in <module>
    from distutils.version import LooseVersion
ModuleNotFoundError: No module named 'distutils'
[6/[26](https://github.com/jimrhoskins/zmk-config/actions/runs/13021471303/job/36322849955#step:11:27)5] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /tmp/tmp.BG25H5oV21
Error: Process completed with exit code 1.
```

I am not sure how best to proceed. I haven't changed anything, and my config is not that novel (I think), so I would imagine this is happening to others, but I cannot find any evidence. 

Is this a wider issue, or do I need to restart from scratch, or import/modify the workflow to ensure python-distutils is enabled.

caksoylar, 2025-01-29:
I think your repo is private, can you make it public?

jimrhoskins, 2025-01-29:
@caksoylar Whoops, my mistake, It is public now. Thanks

caksoylar, 2025-01-29:
It looks like you are using a *really* old ZMK branch: https://github.com/jimrhoskins/zmk-config/blob/main/config/west.yml#L13. I guess something with the Docker image, reusable workflow and that revision isn't playing well right now.

(Mouse keys are now [available officially](https://zmk.dev/docs/keymaps/behaviors/mouse-emulation), so you shouldn't have too much trouble following the linked documentation and switch to `zmkfirmware@main`.)

If you update that, I'd recommend you update the reusable workflow revision to `main` as well: https://github.com/jimrhoskins/zmk-config/blob/main/.github/workflows/build.yml

Note that as usual, `main` can break your config occasionally, though these breakages are announced on Discord and [blog posts](https://zmk.dev/blog). ZMK actually started [tagging releases](https://github.com/zmkfirmware/zmk/releases) so feel free to pin to that in your west.yml instead. However there is no release yet that contains pointing features.

jimrhoskins, 2025-01-29:
Oh! Of course. It's been a while since I worked on this, and completely forgot that I had used that branch.. Thanks for pointing that out, I can probably work it out from here.

Thanks for taking the time to look!

closed 2025-01-29 by jimrhoskins

jimrhoskins, 2025-01-29:
@caksoylar Thanks for your help unsticking me. I got back onto the main repo and updated to the native mouse commands and everything works great. Really appreciate it!

caksoylar, 2025-01-29:
Glad you got everything working well!