zmkfirmware/zmk #2805: Extra Shift Activation with Mod Morph

Namarimizu, 2025-02-01:
This issue may be particularly problematic for users who type in Chinese, as many Chinese input methods use a single left shift press as a Chinese-English language toggle.

I am trying to replicate urob's timeless home-row mods using `mod-morph`. Specifically, I want to configure:
- `shift + ,` → `;`
- `shift + .` → `:`

However, while the second mapping (`shift + . = :`) works as expected, the first mapping (`shift + , = ;`) triggers an additional shift tap before the semicolon.

Because of this, every time I try to input `;`, it misfires and switches input modes unexpectedly.

I suspect this happens because `:` inherently requires `shift` to be triggered, so the single `shift` press is effectively “consumed” in the process. However, since `;` does not require `shift`, the `mod-morph` behavior results in first pressing `shift`, then `;`, even though I have not configured any `keep-mods` settings.

Any insights or potential fixes would be greatly appreciated!

urob, 2025-02-05:
Unfortunately, flashing mods is a side effect of how mod-morph works. This is because when you press shift + comma, there is no way for the mod-morph to retro-actively untap shift. The best it can do is to release shift the moment you press comma (ie, the moment you press the mod-morph).

Namarimizu, 2025-02-05:
> Unfortunately, flashing mods is a side effect of how mod-morph works. This is because when you press shift + comma, there is no way for the mod-morph to retro-actively untap shift. The best it can do is to release shift the moment you press comma (ie, the moment you press the mod-morph).

Thanks for your explanation, and also for your amazing Timeless HRMs!

closed 2025-02-05 by Namarimizu