zmkfirmware/zmk #2829: Sofie NRF52840 Initial config build fails for both halves

bistace, 2025-02-20:
Hello,

I followed the instructions in the docs for my Sofle using NRF52840 but the initial config build in Github Actions fails for both halves. The workflows fail during the build step with the error:
```
devicetree error: /tmp/zmk-config/zmk/app/boards/shields/sofle/sofle.dtsi:75 (column 1): parse error: undefined node label 'pro_micro_i2c'
```

This is the default keymap and I have not modified it, could you please help me solving this issue?

Here is the complete log for one of the halves:

```
Run west build -s zmk/app -d "/tmp/tmp.5SOx5vWMtM" -b "nrf52840dk_nrf52840"  -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD="sofle_left" -DZMK_EXTRA_MODULES='/__w/zmk-config/zmk-config' 
  west build -s zmk/app -d "/tmp/tmp.5SOx5vWMtM" -b "nrf52840dk_nrf52840"  -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD="sofle_left" -DZMK_EXTRA_MODULES='/__w/zmk-config/zmk-config' 
  shell: sh -x {0}
  env:
    build_dir: /tmp/tmp.5SOx5vWMtM
    base_dir: /tmp/zmk-config
    zephyr_version: 3.5.0
    extra_west_args: 
    extra_cmake_args: -DSHIELD="sofle_left" -DZMK_EXTRA_MODULES='/__w/zmk-config/zmk-config'
    display_name: sofle_left - nrf52840dk_nrf52840
    artifact_name: sofle_left-nrf52840dk_nrf52840-zmk
+ west build -s zmk/app -d /tmp/tmp.5SOx5vWMtM -b nrf52840dk_nrf52840 -- -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD=sofle_left -DZMK_EXTRA_MODULES=/__w/zmk-config/zmk-config
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /tmp/zmk-config/zmk/app
-- CMake version: 3.30.0
-- Found Python3: /usr/bin/python3 (found suitable version "3.12.3", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /tmp/zmk-config/zephyr/.cache
-- Zephyr version: 3.5.0 (/tmp/zmk-config/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Adding /tmp/zmk-config/zmk/app/boards/shields/sofle
-- ZMK Config directory: /tmp/zmk-config/config
-- ZMK Config Kconfig: /tmp/zmk-config/config/sofle.conf
-- Using keymap file: /tmp/zmk-config/config/sofle.keymap
-- Using keymap file: /tmp/zmk-config/config/sofle.keymap
-- Board: nrf52840dk_nrf52840
-- Shield(s): sofle_left
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found toolchain: zephyr 0.16.3 (/opt/zephyr-sdk-0.16.3)
-- Found Dtc: /opt/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /tmp/zmk-config/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: /tmp/zmk-config/zmk/app/boards/shields/sofle/sofle_left.overlay
-- Found devicetree overlay: /tmp/zmk-config/config/sofle.keymap
devicetree error: /tmp/zmk-config/zmk/app/boards/shields/sofle/sofle.dtsi:75 (column 1): parse error: undefined node label 'pro_micro_i2c'
-- In: /tmp/tmp.5SOx5vWMtM/zephyr, command: /usr/bin/python3;/tmp/zmk-config/zephyr/scripts/dts/gen_defines.py;--dts;/tmp/tmp.5SOx5vWMtM/zephyr/zephyr.dts.pre;--dtc-flags;'-Wno-unique_unit_address_if_enabled';--bindings-dirs;/tmp/zmk-config/zmk/app/module/dts/bindings;/tmp/zmk-config/zmk/app/dts/bindings;/tmp/zmk-config/zephyr/dts/bindings;--header-out;/tmp/tmp.5SOx5vWMtM/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/tmp/tmp.5SOx5vWMtM/zephyr/zephyr.dts.new;--edt-pickle-out;/tmp/tmp.5SOx5vWMtM/zephyr/edt.pickle;--vendor-prefixes;/tmp/zmk-config/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/tmp/zmk-config/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /tmp/zmk-config/zephyr/cmake/modules/dts.cmake:279 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /tmp/zmk-config/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /tmp/zmk-config/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/tmp.5SOx5vWMtM -GNinja -DBOARD=nrf52840dk_nrf52840 -DZMK_CONFIG=/tmp/zmk-config/config -DSHIELD=sofle_left -DZMK_EXTRA_MODULES=/__w/zmk-config/zmk-config -S/tmp/zmk-config/zmk/app
Error: Process completed with exit code 1.
```

Nick-Munnich, 2025-02-20:
https://www.nordicsemi.com/Products/Development-hardware/nRF52840-DK 
This is the board known as nrf52840dk_nrf52840.
Most likely you have a bad aliexpress clone of the nice nano, which you should be using as the board if so. A word of caution: those clones are unreliable and may unexpectedly fail in the future, if they don't have any flaws already.

bistace, 2025-02-20:
Hi @Nick-Munnich and thank you for your reply. I am surely missing something but my issue is with the base GitHub actions that was configured by zmk-config. In the zmk-config configuration prompt, I indicated a Sofle with NRF52840 and the initial Github Actions workflow is failing without me touching anything.

caksoylar, 2025-02-20:
>  In the zmk-config configuration prompt, I indicated a Sofle with NRF52840

You have chosen `nrf52840dk_nrf52840` in the prompt for the board, which is what Nick linked to. As you can see, its shape/pinout is not compatible to be installed in a Sofle.

"NRF52840" by itself doesn't mean much: It is the name of the chip, not the board. There are many boards using that chip. Our assumption is you are using one of the first two [here](https://github.com/joric/nrfmicro/wiki/Alternatives), which means you need to select nice!nano v2 instead in the setup script. (TBH, I'd rather people take it up with the seller if they aren't clearer on what controller _board_ they are selling to you, rather than community compensating for it.)

bistace, 2025-02-20:
Ok I see, thanks a lot for your help!