zmkfirmware/zmk #2830: Building Custom shield error  -CMake Error: Invalid SHIELD;

Jogrefer, 2025-02-21:
**[> My repo <](https://github.com/Jogrefer/zmk-config-jogrefer)**  |  **[Build Error](https://github.com/Jogrefer/zmk-config-jogrefer/actions/runs/13446719420/job/37573567516)**

--

I tried tutorials and read documentation using Github Actions and I haven't could build the firmware.
I'm doing a handwired from scratch prototype, following tutorials it seems like the physical step is not important at all becuse it's not complicated. 

I don't know if I am doing something wrong 🤷🏽‍♂️

caksoylar, 2025-02-21:
Please make sure to follow the documentation (and only the documentation, not LLM-generated code etc.) at https://zmk.dev/docs/development/new-shield carefully.

There are so many issues here that it would be better to start from scratch rather than trying to patch it. This is just a few of the obvious issues I see:
- Your repo isn't set up from the template as instructed, as a module, and `boards/shields/jogrefer` shouldn't be under `config/`
- `Kconfig.defconfig` and `Kconfig.shield` should be capitalized exactly like this
- You should only have `west.yml`, `jogrefer.keymap` and `jogrefer.conf` under `config/`
- [These](https://github.com/Jogrefer/zmk-config-jogrefer/blob/main/config/boards/shields/jogrefer/jogrefer.overlay#L17) are not how you specify GPIO pins
- You shouldn't have a [space after comma](https://github.com/Jogrefer/zmk-config-jogrefer/blob/main/config/boards/shields/jogrefer/kconfig.shield#L2)
- I'd also suggest you start with a minimal keymap that works, then add things -- it currently has many issues

If you follow the new shield guide carefully, you shouldn't have above issues.

caksoylar, 2025-02-27:
Closing due to user-related issue.

closed 2025-02-27 by caksoylar