zmkfirmware/zmk #2832: [Feature Request] Can I send IR signals from a ZMK keyboard?

myst729, 2025-02-24:
I have an HTPC connected to my TV. So I handcrafted a tiny-sized keyboard to control the PC. But I also would like to watch TV shows over the cable signal. Then I have to grab the TV remote to switch the HDMI source between cable and PC. So the idea is to integrate an IR LED on the keyboard and send signals like the TV remote.

caksoylar, 2025-02-25:
Do TV remotes use the HID protocol like USB or BLE? If not, I doubt this is feasible to add to ZMK. If so, it could technically be another endpoint (in addition to USB and BLE) but I'd still doubt if the development+maintenance would be worth it. It could be feasible to make it a module that listens to keycode events and sends it over the different endpoint.

myst729, 2025-02-26:
Well... no. Probably I should turn to Zephyr to see what they have, and include the relevant code into my zmk-config, leaving zmkfirmware intact. Good point.