zmkfirmware/zmk #2841: Delay when using a momentary layer switch when external power is off

bistace, 2025-02-26:
Hello,

when external power is off, if I activate a momentary layer (&mo) and immediately press a key on this layer , it is not sent. I have to wait a noticeable amount of time between the momentary layer activation and the keypress for it to be sent. This delay is absent when external power is on and the keypress on the activated layer is immediately sent.

To be clearer, here is a small demo of what is happening (let's imagine that on the layer, the letter A sends the number 1):

```
External power on:
&mo Layer activation -> Key press A -> Key press A -> Key press A
Output on screen: 111

External power off:
&mo Layer activation -> Key press A -> Key press A -> Key press A
Output on screen: 1
```

Here is a link to my zmk-config repo if it can be of any help to you but there is nothing fancy inside:
https://github.com/bistace/zmk-config

Could you please help me solve this issue?

Nick-Munnich, 2025-02-26:
Could you verify that this isn't caused by your keyboard falling asleep?
Could you collect some logs https://zmk.dev/docs/development/usb-logging

bistace, 2025-02-26:
same experience bug with external power on: activate the mometary layer on the right half and press 4 times the equivalent of the letter E on a QWERTY keyboard. This time all four key presses were displayed on my screen.

Here are the logs for the left half with external power on:
```
tio -t /dev/tty.usbmodem1101
[13:43:38.225] tio 3.8
[13:43:38.226] Press ctrl-t q to quit
[13:43:38.228] Connected to /dev/tty.usbmodem1101
[13:43:52.491] [00:01:47.249,938] <dbg> zmk: split_central_notify_func: [NOTIFICATION] data 0x2000f20b length 16
[13:43:52.492] [00:01:47.249,969] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.492] [00:01:47.249,969] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.493] [00:01:47.250,000] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.493] [00:01:47.250,000] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.495] [00:01:47.250,030] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.496] [00:01:47.250,030] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.496] [00:01:47.250,061] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.497] [00:01:47.250,061] <dbg> zmk: split_central_notify_func: data: 2
[13:43:52.498] [00:01:47.250,091] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.498] [00:01:47.250,091] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.499] [00:01:47.250,122] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.507] [00:01:47.250,122] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.507] [00:01:47.250,152] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.508] [00:01:47.250,152] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.509] [00:01:47.250,183] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.510] [00:01:47.250,183] <dbg> zmk: split_central_notify_func: data: 0
[13:43:52.511] [00:01:47.250,274] <dbg> zmk: peripheral_event_work_callback: Trigger key position state change for 57
[13:43:52.512] [00:01:47.250,366] <dbg> zmk: zmk_keymap_apply_position_state: layer_id: 0 position: 57, binding name: momentary_layer
[13:43:52.513] [00:01:47.250,427] <dbg> zmk: mo_keymap_binding_pressed: position 57 layer 2
[13:43:52.514] [00:01:47.250,457] <dbg> zmk: set_layer_state: layer_changed: layer 2 state 1
[13:43:52.514] [00:01:47.251,220] <dbg> zmk: vddh_sample_fetch: ADC raw 3364 ~ 4925 mV => 100%
[13:43:52.712] [00:01:47.388,153] <dbg> zmk: kscan_matrix_read: Sending event at 1,3 state on
[13:43:52.712] [00:01:47.388,244] <dbg> zmk: zmk_physical_layouts_kscan_process_msgq: Row: 1, col: 3, position: 15, pressed: true
[13:43:52.713] [00:01:47.388,305] <dbg> zmk: zmk_keymap_apply_position_state: layer_id: 2 position: 15, binding name: key_press
[13:43:52.715] [00:01:47.388,336] <dbg> zmk: on_keymap_binding_pressed: position 15 keycode 0x70022
[13:43:52.716] [00:01:47.388,366] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x22 implicit_mods 0x00 explicit_mods 0x00
[13:43:52.717] [00:01:47.388,397] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[13:43:52.718] [00:01:47.388,397] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[13:43:52.719] [00:01:47.471,191] <dbg> zmk: kscan_matrix_read: Sending event at 1,3 state off
[13:43:52.719] [00:01:47.471,343] <dbg> zmk: zmk_physical_layouts_kscan_process_msgq: Row: 1, col: 3, position: 15, pressed: false
[13:43:52.720] [00:01:47.471,405] <dbg> zmk: zmk_keymap_apply_position_state: layer_id: 2 position: 15, binding name: key_press
[13:43:52.721] [00:01:47.471,435] <dbg> zmk: on_keymap_binding_released: position 15 keycode 0x70022
[13:43:52.723] [00:01:47.471,466] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x22 implicit_mods 0x00 explicit_mods 0x00
[13:43:52.724] [00:01:47.471,466] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[13:43:52.725] [00:01:47.471,496] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[13:43:52.875] [00:01:47.552,154] <dbg> zmk: kscan_matrix_read: Sending event at 1,3 state on
[13:43:52.875] [00:01:47.552,246] <dbg> zmk: zmk_physical_layouts_kscan_process_msgq: Row: 1, col: 3, position: 15, pressed: true
[13:43:52.876] [00:01:47.552,307] <dbg> zmk: zmk_keymap_apply_position_state: layer_id: 2 position: 15, binding name: key_press
[13:43:52.877] [00:01:47.552,337] <dbg> zmk: on_keymap_binding_pressed: position 15 keycode 0x70022
[13:43:52.878] [00:01:47.552,368] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x22 implicit_mods 0x00 explicit_mods 0x00
[13:43:52.880] [00:01:47.552,398] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[13:43:52.881] [00:01:47.552,398] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[13:43:52.882] [00:01:47.634,155] <dbg> zmk: kscan_matrix_read: Sending event at 1,3 state off
[13:43:52.882] [00:01:47.634,277] <dbg> zmk: zmk_physical_layouts_kscan_process_msgq: Row: 1, col: 3, position: 15, pressed: false
[13:43:52.884] [00:01:47.634,338] <dbg> zmk: zmk_keymap_apply_position_state: layer_id: 2 position: 15, binding name: key_press
[13:43:52.884] [00:01:47.634,368] <dbg> zmk: on_keymap_binding_released: position 15 keycode 0x70022
[13:43:52.885] [00:01:47.634,399] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x22 implicit_mods 0x00 explicit_mods 0x00
[13:43:52.887] [00:01:47.634,429] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[13:43:52.888] [00:01:47.634,429] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[... cut ...]
[13:43:52.583] [00:01:09.276,275] <dbg> zmk: zmk_physical_layouts_kscan_process_msgq: Row: 4, col: 3, position: 57, pressed: true
[13:43:52.585] [00:01:09.276,306] <dbg> zmk: split_listener: 
[13:43:52.585] [00:01:09.277,374] <dbg> zmk: vddh_sample_fetch: ADC raw 3003 ~ 4395 mV => 100%
[13:43:53.501] [00:01:10.194,152] <dbg> zmk: kscan_matrix_read: Sending event at 4,3 state off
[13:43:53.501] [00:01:10.194,274] <dbg> zmk: zmk_physical_layouts_kscan_process_msgq: Row: 4, col: 3, position: 57, pressed: false
[13:43:53.502] [00:01:10.194,305] <dbg> zmk: split_listener: 

[13:44:01.303] Disconnected
```

Nick-Munnich, 2025-02-26:
How interesting, the missing key presses aren't even showing up in logging. Could you try disabling widgets one by one and see if one of them fixes it? I'll leave this for someone else to investigate aside from that; I do not have the necessary hardware to poke around here.

bistace, 2025-02-26:
That was it! Setting `CONFIG_ZMK_DISPLAY_WORK_QUEUE_DEDICATED=y` in my config removed the delay when activating a momentary layer when external power is off. Thank you so much for helping me with this @Nick-Munnich and @lesshonor