zmkfirmware/zmk #2870: Is there a way to rapidly press and release a key? Such as a autoclicker, but also for other keys.

yuannan, 2025-03-11:
This feature has been in QMK for a while now, you can rapidly send over new keys with very good consistency using the deferred execution API.

I don't see an easy way to do this on ZMK, if anyone knows how please let me know.

I think it would be nice to have a behaviour such as `&turbo` that can then take another behaviour such as `&kp` or `&gresc` and then rapidly sends the keys.

I think it would also be nice to able to modify the send delay with other keys, increasing or decreasing the delay between each send. It might also be nice to have a toggle version of `&turbo` where it will keep sending keys until it is pressed again. This might be fairly trivial if you just overload `&turbo` with `&kt`.

FYI: I'm not using this to cheat in video games or anything like that, something you do need an autoclicker for poorly designed GUIs that don't have drag functionally.

zhangweidai, 2025-03-16:
I need this too

urob, 2025-03-17:
Have you seen https://github.com/zmkfirmware/zmk/pull/1414?

caksoylar, 2025-03-17:
Also might be relevant: https://github.com/badjeff/zmk-behavior-key-tempo

yuannan, 2025-03-20:
Anyone know why this hasn't been merged? I'd be happy to help out if there is a clear spec for this PR to meet.