zmkfirmware/zmk #2885: Feature Request: Allow switching between central and peripheral mode at run time or boot time

keks, 2025-03-19:
I don't know whether this is feasible with the current software architecture, but it would be nice if I could have my keyboard in peripheral mode at my desk (using a dongle), but be able to switch it to central mode so I don't need a dongle. This would be handy while on the go, or when connecting to devices with no/few USB ports, such as phones or tablets.

If this is possible already, sorry for the noise! But I couldn't find it in the docs, and the wording of the dongle page made it sound like it is not (under Disadvantages: "The keyboard becomes unusable without the dongle.")

Ririshi, 2026-01-07:
I would love to have this feature for the same reason as OP. If it was possible to select a mode on boot, one could set up one of the keyboard halves to first boot into peripheral mode and automatically reboot to central mode if it doesn't find a central dongle to connect to within a preset timeout. Actually setting the boot mode could be a setting written to EEPROM? I'm not knowledgeable enough to add much more of use here.

qriaa, 2026-05-25:
This is the only thing that prevents me from building my own dongle, this seems absolutely IMMENSELY useful.
<sub>_Offtopic: Also having two dongles with dynamic switching between them would be cool_<sub>