zmkfirmware/zmk #2905: &kp modifiers (LA, LS, etc.) are released before modified key is released resulting in strange program behavior

deggua, 2025-04-03:
I recently flashed ZMK to my corne and setup a config that matches my QMK config. I noticed some weird behavior using VSCode where it seemed like I was hitting LALT on its own (menu bar gets selected with _ under appropriate letters for hotkey navigation, which would screw up subsequent inputs) when using certain modified keys in my keymap (`&kp LA(UP)`, `&kp LA(LEFT)`, etc.). This never occurred on my QMK board. Using a keyboard testing program (Switch Hitter), the difference between QMK and ZMK behavior for modified keys (`&kp LA(x)` vs `SS_LALT(SS_TAP(x))`) seems to be the associated modifier release order:

ZMK Keymap: `&kp LA(LEFT)`
```
41:30.0634 LAlt (0x12, BIOS 0x38) DOWN
41:30.0639 Left (0x25, BIOS 0xE04B) DOWN
41:30.0739 LAlt (0x12, BIOS 0x38) UP -> 105ms
41:30.0743 Left (0x25, BIOS 0xE04B) UP -> 104ms
```

QMK Keymap: `SS_LALT(SS_TAP(X_LEFT))`
```
42:11.0453 LAlt (0x12, BIOS 0x38) DOWN
42:11.0458 Left (0x25, BIOS 0xE04B) DOWN
42:11.0463 Left (0x25, BIOS 0xE04B) UP -> 5ms
42:11.0468 LAlt (0x12, BIOS 0x38) UP -> 19ms
```

This seems to lead to odd behavior in some environments. I'm guessing I could workaround this with a macro for each key I am currently using `&kp MOD(KEY)` for, but this is inconvenient and the key release ordering doesn't seem correct to me. I would expect that the mod would be pressed before the key itself and released after the key is released.

ZMK keymap: https://github.com/deggua/zmk-config/blob/master/config/corne.keymap
QMK keymap: https://github.com/deggua/qmk_firmware/blob/master/keyboards/crkbd/keymaps/deggua/keymap.c

caksoylar, 2025-04-04:
It looks like you are using a macro in QMK, rather than modifier functions like in ZMK. QMK equivalent would be `LALT(KC_UP)`.

You can similarly try creating [macros](https://zmk.dev/docs/keymaps/behaviors/macros) in ZMK to tweak the timings as you like, and they can even be parametrized so you only need to define one and use with keycode parameters.

Another solution might be to enable [`CONFIG_ZMK_HID_SEPARATE_MOD_RELEASE_REPORT`](https://zmk.dev/docs/config/system#hid), which could help or not.

deggua, 2025-04-05:
> Another solution might be to enable [`CONFIG_ZMK_HID_SEPARATE_MOD_RELEASE_REPORT`](https://zmk.dev/docs/config/system#hid), which could help or not.

This seems to produce the desired behavior without having to write a macro:
```
10:54.0882 LAlt (0x12, BIOS 0x38) DOWN
10:54.0887 Up (0x26, BIOS 0xE048) DOWN
10:54.0941 Up (0x26, BIOS 0xE048) UP -> 54ms
10:54.0945 LAlt (0x12, BIOS 0x38) UP -> 65ms
```

Thanks for pointing me to that config option. I assume this isn't a bug then, though the default does seem a bit intuitive imo.

closed 2025-04-05 by deggua