zmkfirmware/zmk #2939: Feature Request: Japanese Duplex Matrix kscan

andre-tm-hui, 2025-05-09:
Wondering if anyone has considered implementing this before, and if there's any reason not to do this. This came up since I recently got a Keyball61 kit, thinking ZMK would slot in no problem, only to find out it definitely doesn't.

What this essentially entails is a matrix that can be scanned both ways (`col2row` and `row2col` simultaneously), effectively doubling the number of switches per controller.

For more info, see [here](https://kbd.news/The-Japanese-duplex-matrix-1391.html)

Would appreciate any thoughts/suggestions - I'm not too familiar with the inner workings of ZMK, but thinking of starting by looking at the existing code behind `kscan-gpio-matrix`. Thanks!

caksoylar, 2025-05-09:
I am not 100% on the detail differences of the setups, but I know naming has been confusing for these sorts of setups. Have you checked whether https://zmk.dev/docs/config/kscan#charlieplex-driver works?

fivemeyestore, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

andre-tm-hui, 2025-05-10:
The charlieplex driver works perfectly! Closing since this ticket is redundant now. Thanks @caksoylar

closed 2025-05-10 by andre-tm-hui