zmkfirmware/zmk #2941: Mouse functionality not working with Seeeduino XIAO BLE (nRF52840) controllers

willnapier, 2025-05-13:
## Description
Mouse functionality (motion and scroll) does not work when using Seeeduino XIAO BLE (nRF52840) controllers, while all other keyboard functions work properly. This issue persists across multiple builds, configuration approaches, and connection methods (both USB and Bluetooth).

## Hardware Details
- Keyboard: Totem split ergonomic (38-key)
- Controllers: Seeeduino XIAO BLE (nRF52840)
- Host: macOS Sonoma
- Layout: Miryoku ZMK

## Testing Performed
I've conducted extensive testing to isolate the issue:

1. **Minimal Working Configuration:**
   - Built a minimal Miryoku setup with mouse keys disabled
   - Result: Full Bluetooth and USB functionality works perfectly

2. **Mouse Keys Enabled (Main Branch):**
   - Added `CONFIG_ZMK_MOUSE=y` and `#define MIRYOKU_KLUDGE_MOUSEKEYSPR`
   - Result: Keyboard connects via Bluetooth and USB, but mouse movements/scrolling do not function
   - Keys that should trigger mouse movements show numbers instead (fallback behavior)

3. **Mouse Keys PR Implementation:**
   - Tested PR #101 from manna-harbour/miryoku_zmk
   - Result: Same behavior - keyboard connects but mouse functionality doesn't work

4. **HID Descriptor Refresh:**
   - Followed complete settings reset procedure (including pressing keys after flashing)
   - Cleared all Bluetooth bonds on keyboard and host
   - Re-paired from scratch
   - Result: No change in behavior

5. **USB-Only Testing:**
   - Disabled Bluetooth to test mouse functionality over USB only
   - Result: Mouse keys still don't function over USB

## Configuration Details
Mouse functionality was enabled using:

// In config
CONFIG_ZMK_MOUSE=y
CONFIG_ZMK_SPLIT=y
CONFIG_BT_CTLR_TX_PWR_PLUS_2=y

// In custom_config.h
#define MIRYOKU_KLUDGE_MOUSEKEYSPR
#define ZMK_POINTING_DEFAULT_MOVE_VAL 600
#define ZMK_POINTING_DEFAULT_SCRL_VAL 10

## Additional Notes
- No error messages appear in USB logging
- All other ZMK functionality works perfectly (layers, modifiers, Bluetooth profiles)
- The issue appears specific to XIAO BLE controllers - research suggests nice!nano controllers work with mouse keys
- This appears to be a hardware compatibility issue rather than a configuration problem

I'm happy to provide any additional information or testing that might help resolve this issue.

nmunnich, 2025-05-13:
`CONFIG_ZMK_MOUSE` does not exist. The flag is called `CONFIG_ZMK_POINTING`

willnapier, 2025-05-13:
Thank you for your input, @nmunnich. I checked my configuration files and discovered that both flags were actually already enabled in my original setup:
In original totem.conf
CONFIG_ZMK_MOUSE=y
CONFIG_ZMK_POINTING=y
CONFIG_ZMK_POINTING_SMOOTH_SCROLLING=y

I tried rebuilding with only `CONFIG_ZMK_POINTING=y` (removing the deprecated `CONFIG_ZMK_MOUSE`), but unfortunately, the mouse functionality still doesn't work. The keyboard connects fine via both Bluetooth and USB, all other keys work, but mouse movements and scrolling don't function.

This suggests the issue is deeper than just the configuration flag naming. I'm using Seeeduino XIAO BLE controllers, and I wonder if there might be hardware-specific compatibility issues with the ZMK pointing implementation on these controllers.

Has anyone successfully used mouse/pointing features with XIAO BLE controllers? Any suggestions for further troubleshooting would be greatly appreciated.

caksoylar, 2025-05-13:
Please share a link to the build. This is very likely to be a configuration issue, either with the Miryoku feature flag or how it is used.

The issue I know right now is `CONFIG_ZMK_POINTING_SMOOTH_SCROLLING` not working properly with Windows over BLE, causing scrolling to be too fast. Everything else should be working, especially over USB (people can have trouble with BLE due to not [refreshing the HID descriptor](https://zmk.dev/docs/features/bluetooth#refreshing-the-hid-descriptor)).

(FWIW I am using two keyboards with Xiao BLEs, running latest ZMK and mouse emulation working.)

willnapier, 2025-05-13:
@caksoylar This is very helpful and encouraging, especially that you have Xiao BLEs working with mouse emulation. Would you be willing to share your config?

I have set up a github repo specifically to share the config I have tried:

https://github.com/willnapier/totem-miryoku-mouse

caksoylar, 2025-05-13:
My config is [here](https://github.com/caksoylar/zmk-config), but there isn't anything fancy needed to enable mouse keys. Just
- add `CONFIG_ZMK_POINTING=y` to a config file used by the build (mine is in `config/default.conf`)
- use `&mmv`/`&msc` in your keymap and press them]

Your repo is very nicely documented, though it's not a ZMK config repo we can inspect build results in. I'll try to reproduce locally following your build commands later today.

caksoylar, 2025-05-14:
FYI I tried to follow up on Discord since I noticed you already had an open forum thread: https://discord.com/channels/719497620560543766/1371296568615112805 Given this is a user issue and there is a better avenue to continue the discussion, I am closing this issue.