zmkfirmware/zmk #2944: I want to add a mouse scroll wheel encoder to my keyboard, but I'm currently encountering some issues

njian-github, 2025-05-17:
I want to add a mouse scroll wheel encoder to my keyboard, but I'm currently experiencing some issues. I've configured the two scroll directions to output 0 and 1 respectively. However, when I scroll the encoder in one direction, both 0 and 1 sometimes get triggered. Normally, scrolling in one direction should only output either 0 or 1. I’d like to ask if there is a way to fix this issue. Below is my configuration file. Please help me. Thank you!
`    right_encoder: encoder_right {
        compatible = "alps,ec11";
        a-gpios = <&pro_micro 3 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
        b-gpios = <&pro_micro 2 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
        steps = <12>;
        status = "okay";
    };

    sensors: sensors {
        compatible = "zmk,keymap-sensors";
        sensors = <&right_encoder>;
        triggers-per-rotation = <24>;
    };
`
`    keymap {
        compatible = "zmk,keymap";

        default_layer {
            bindings = <&kp A
                        &kp A
                        &kp A
                        &kp A
                        &kp A
                        &bt BT_CLR_ALL
                        &bootloader
                        &out OUT_BLE
                        &sys_reset
                        &studio_unlock>;
            sensor-bindings = <&inc_dec_kp N1 N0>;
        };
        
    };`

[image]

caksoylar, 2025-05-22:
This was discussed on Discord, it is likely that it is a resolution setting issue. Recommendation was to try `triggers-per-rotation = <24>;` with either `steps = <48>;` or `steps = <96>;`. Closing, please follow up on the Discord thread.

njian-github, 2025-05-23:
> 这在 Discord 上讨论过，很可能是分辨率设置问题。建议尝试使用 或 。关闭，请跟进 Discord 帖子。`triggers-per-rotation = <24>;``steps = <48>;``steps = <96>;`

Thank you! The problem has been solved.