zmkfirmware/zmk #2958: Request for Permission to Integrate Zephyr PF1550 PMIC Drivers into ZMK

BionicCode, 2025-06-07:
Hello ZMK maintainers,

I would like to request permission to integrate the NXP PF1550 PMIC support drivers from upstream Zephyr into the ZMK codebase. The PF1550 provides charger, regulator (buck/LDO), and power management features that would be valuable for custom hardware using ZMK.

Specifically, I propose to:

Integrate the Zephyr drivers found in:
drivers/mfd/mfd_pf1550.c
drivers/charger/charger_pf1550.c
drivers/regulator/regulator_pf1550.c
Add any necessary Kconfig, DTS, and CMake updates to support these drivers within ZMK’s build system.
Ensure licensing and attribution are consistent with Zephyr’s Apache 2.0 license.
Would the ZMK team be open to this integration, and are there any requirements or constraints I should be aware of before initiating a PR?

Thank you!

nmunnich, 2025-06-07:
After the [wired split](https://github.com/zmkfirmware/zmk/pull/2886) feature is finalised the next major feature on the roadmap is updating ZMK to use Zephyr 4.1. Anything from upstream should go into our Zephyr fork and definitely not into ZMK itself.

BionicCode, 2025-06-07:
@nmunnich 
That's great. It's so nice to witness this project growing and maturing. I wish I could contribute more. 

Regarding the driver support, I'm really confused. Because the drivers for the FUSB302 and PF1550 are supported on Zephy since v2.1 and v3. So I was assuming that you kind of hand pick supported hardware. 
Isn't the current ZMK version using Zephyr 3.5? So these drivers should be available, right?

Because I had a discussion on Discord (I even think it was you) when I was asking for support of the fuel gauge or the nPM1300 PMIC in general. I was then told that this device is currently not supported (whereas the drivers were added in Zephyr 3.5). 

From the _west.yml_ I can tell that ZMK is on top of Zephyr 3.5. 
Can you please confirm that all drivers that were added to Zephyr 3.5 or earlier (including FUSB302, PF1550, nPM1300 in particular) are 100% available from the current ZMK context?

petejohanson, 2025-06-07:
There's a difference between "a driver is available for the hardware" and "client code to consume the particular subsystem API exists"

Feel free to check the Zephyr 3.5 release for those drivers, and then you'd need to look at how the ZMK should consume them.

BionicCode, 2025-06-07:
Maybe I should mention that I'm not using a predefined board or shield. I'm creating a custom onboard solution that uses the nRF52840 as MCU.

petejohanson, 2025-06-07:
> Maybe I should mention that I'm not using a predefined board or shield. I'm creating a custom onboard solution that uses the nRF52840 as MCU.
> 

That doesn't really make a difference, as far as any ZMK code to integrate these is concerned.

BionicCode, 2025-06-07:
Ok. I have just double checked the Zephyr submodule and it actually contains a tiny tiny fraction of the original drivers, where the nPM1300 seems to be made fully available.

Does that mean that you will stick to the restrictive supported driver policy or will be all Zephyr drivers available from the v4.1 upgrade?
Do you manually delete these drivers? If so is there a chance you will be interested to add support for the FUSB302  and PF1550 during the next upgrade?
I guess there is a good reason for the removal of the majority of drivers. I hope the conditions have changed and drivers will be available.

I'm very willing to help if help is needed to make this all work (including the upgrade).

petejohanson, 2025-06-07:
> > The drivers are not removed, or in any way restricted. Anyone can choose to write code to consume some other subsystem (e.g. fuel gauge, or USB power negotiation) as they see fit to supplement ZMK.
> 
> 
> 
> I see. Thank you very much for the clarification.  I have checked the Zephyr history and just found that my claims about the versions were not correct. The drivers of interest were just recently added  to v4.1 and later to Zephyr.  Too late I'm afraid.
> 
> The FUSB307 (a very nice device) was just added a few 2 weeks ago. While  the device is already integrated in the current Zephyr, it is not included in the latest release. It should be available in the next release for sure (v4.2). Too bad. Looks like I have to look for an alternative device then.
> 
> The PF1550 is available from v4.1. So we will be able to use it after the upgrade.
> I have to apologize for the confusion.

Yup, makes sense.

> 
> Will you upgrade at a higher pace in future? I understand that this involves al lot of work as API become deprecated or have breaking changes. Just asking out of curiosity.

I don't have anything definitive yet. We've started doing tagged ZMK releases so we can more easily do breaking changes (like 4.1 will introduce), but that doesn't change the large effort required to do Zephyr upgrades. I have some ideas for how to improve this, but no promises at this point.
> 
> Do you have any idea when the next upgrade could happen? I'm in the middle of development so I'm not too much in a hurry (I'm just saying. I know it's not relevant).

I hope to get the split work merged in ~2 weeks, and will hopefully have preliminary 4.1 branch up in a week or so for bleeding edge testers.

The actual merge of that will involve a ton of testing, docs, blog post, etc. probably 2 months out, give or take.

> Again, if i can do anything to help you progress just let me know.  I'm not a split keyboard user so help with the split keyboard PR #2886 that nmunnich has mentioned is very  limited. But let me know if I can help. I will then try my best.

BionicCode, 2025-06-07:
> The actual merge of that will involve a ton of testing, docs, blog post, etc. probably 2 months out, give or take.

Wow. Respect. Are you doing this alone (the major part I guess)? That's a huge agenda for one person.
I really fell bad for leeching your work...  I hope you don't feel like I'm unpolite and demanding. I'm really not. I'm a software developer myself and I understand the efforts it takes.

Do you think I could manually add the driver files to bridge the time until the next release? I have a felling that it's not that trivial. It really depends on the architecture of Zephyr. 

Maybe you could create a task list of the tasks you have in order to help you with the next release. I could then pick something of what I think I can complete.

THANKS FOR YOUR WORK!!! This is a great modern firmware you have created here. Much respect! The idea to set it on top of Zephyr is really smart. Although at the same time you have created too much pressure to keep your project up to date. You need more developers. Maybe you can connect with Zephyr to assist you? i don't know, I just wish I could be useful ;)

closed 2025-06-08 by BionicCode