zmkfirmware/zmk #2959: Request for Permission to Integrate Zephyr FUSB302 USB Type-C/PD Controller Driver into ZMK

BionicCode, 2025-06-07:
Hello ZMK maintainers,

I would like to request permission to integrate support for the FUSB302 USB Type-C/USB Power Delivery controller into the ZMK codebase by leveraging the existing Zephyr driver (drivers/usb/usb_c_tcpc/fusb302.c).

The FUSB302 is a common solution for USB Type-C and Power Delivery negotiation and is useful for keyboards and other input devices that may need to negotiate higher current or specific roles when connected via USB-C.

It's interesting because this way we can safely support fast charging adapter or to fully utilize the full range of USB specs for charging:

| USB Version              | Max Current (Standard Port) | Notes                                    |
|--------------------------|-----------------------------|------------------------------------------|
| USB 2.0                  | 500 mA                      |                                          |
| USB 3.0/3.1/3.2          | 900 mA                      |                                          |
| USB BC 1.2 CDP/DCP       | 1.5 A                       | Charging ports                           |
| USB Type-C (no PD)       | 3.0 A                       | If supported by source                   |
| USB PD (Type-C/PD)       | 5.0 A                       | Up to 20V, needs negotiation and cable   |

For example, in tandem with the PF1550 PMIC (which I proprsed earlier) we can fully support USB 3 and USB BC (charging port) to allow for fast charging.
The key is that we need to properly negotiate the capabilities of the USB host in order to avoid overloading it.

Specifically, I propose to:

Integrate Zephyr’s FUSB302 driver and any necessary supporting files (Kconfig, DTS bindings, etc.) into ZMK.
Add configuration and documentation for enabling and using the FUSB302 in ZMK-powered devices.
Ensure licensing and attribution are consistent with Zephyr’s licensing.
Are there any requirements, constraints, or concerns the ZMK team would like addressed before I begin work on a pull request for this integration?

Thank you!

petejohanson, 2025-06-07:
You don't need to pull those Zephyr bits into ZMK, just write code to consume them once we are using a new enough Zephyr version that includes them. I've got some *early* 4.1 work that I'll be posting for folks to test "soon" that likely includes those drivers. It's then a matter of writing the supporting code in ZMK.

BionicCode, 2025-06-09:
@petejohanson I think I was a bit confused the other day. Maybe you can help to validate my idea that we can always low-level interact with peripheral devices via bus protocols like I²C/TWI, right? The devices of interest are rather trivial (e.g. not too complex protocols or procedures/state machines) so missing the convenience layer of a driver should be fine. 

So, I just use the nRF52840 driver API to communicate via the configured/declared TWI pins to read/write registers of the peripheral device (e.g. FUSB307 USB-C controller, especially the PF1550 PMIC) to query states and control behavior. I don't need any custom USB negotiations. 

Bottom line: we can basically use every peripheral devices as long as they support the communication bus protocol of the MCU (the device that hosts the OS).

Do you agree or have I missed a crucial point?

BionicCode, 2025-06-14:
@petejohanson Too bad you just forsake me. A brief feed back like "I don't have time for you" would be polite, at least. Pretty disappointing.  Maybe I said something wrong. In this case I have to apologize. English is not my first language so I probably said something  that came over impolitely. 

However, I have now decided to go with QMK as they are more supportive and the flexibility is unmatched - at least at this point in time. 
I really like the concept of ZMK and thought it's worth advertising and utilizing it. But the extremely limited feedback makes me fell not welcome. Additionally the very slow upgrade policy in terms of staying in sync with Zephyr, imposes severe hardware restrictions to an already restrictive environment, which is also extremely discouraging. 

Anyways, I love that project and your commitment. You likely don't care about my opinion, but here I am.
All the best for your project and the people who support you.

Best,

closed 2025-06-14 by BionicCode

petejohanson, 2025-06-14:
I have a full time job, a family of two kids, and a personal life. My time that I have available to give to this project is not unlimited. Feel free to use whatever firmware you want to use, but to disparage the project and my effort because I don't respond when you want me to sucks, and is the kind of behavior that drivers maintainers to leave projects.

BionicCode, 2025-06-14:
Oh, sorry. i never meant to be rude. I just wanted to voice my feelings. just how it felt for me the way this conversation has ended abruptly. 

I did not disparage your effort. I even expressed my respect and how much I value your work. This time and in recent conversations too. I did that too on Discord. I wasn't arselicking. I really meant what I said and haven't changed my mind. 
I 'm just disappointed that you did never react. I understand that this is not your full time occupation, for sure. It just didn't felt nice to me.

On the other hand I was really enthusiastic in joining your team and participate. I just felt like I said something stupid which made you ignore me. 

I still love your project and value your efforts and dedication which gives people the chance to build their custom keyboards. My hirt feelings don't change my views on your project. I'm a software developer myself and I think I have a good understanding of the work this project requires.