zmkfirmware/zmk #2971: zmk init fails with fatal error: malformed manifest

tomchestna, 2025-06-20:
Following the install guide here: https://zmk.dev/docs/user-setup-cli

```
$ git --version
git version 2.48.1
$ python3 --version
Python 3.13.3
$ pipx --version
1.7.1

$ zmk init
If you already have a ZMK config repo, enter its URL here. Otherwise, leave this blank to create a new repo.
Repository URL: https://github.com/tomchestna/zmk-config.git
Enter a directory name (zmk-config.git):
Cloning into 'zmk-config.git'...
remote: Enumerating objects: 13, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (5/5), done.
Receiving objects: 100% (13/13), done.
remote: Total 13 (delta 0), reused 10 (delta 0), pack-reused 0 (from 0)
Initializing west application. This may take a while...
=== Initializing from existing manifest repository config
--- Creating /home/tom/Keyboards/zmk-config.git/.zmk/.west and local configuration file
=== Initialized. Now run "west update" inside /home/tom/Keyboards/zmk-config.git/.zmk.
FATAL ERROR: Malformed manifest file: /home/tom/Keyboards/zmk-config.git/.zmk/config/west.yml
  Schema file: /home/tom/.local/share/pipx/venvs/zmk/lib/python3.13/site-packages/west/manifest-schema.yml
  Hint: Schema validation failed:
 - Key 'default' was not defined. Path: ''.
 ```

Given the reference to west (which I have installed via pipx, not sure if that's relevant):

```
$ west --version
West version: v1.4.0
```

joelspadin, 2025-08-02:
@tomchestna can you verify whether this issue is fixed now?

tomchestna, 2025-08-03:
Ran `pipx reinstall zmk`

```
$ zmk --version
0.3.2
$ west --version
West version: v1.4.0
```

Using the same repository from the initial post still fails.

However, I reran `zmk init` and left the repo blank, going through the flow of setting up a new repository on github. That appeared to work (no errors) and the command concluded with:

```
=== updating zmk (zmk):
HEAD is now at edf5c081 chore(main): release 0.3.0 (#2858)
=== updating nanopb (modules/lib/nanopb):
HEAD is now at 8c60555 Merge pull request #4 from petejohanson/build/find-protoc-fixes
=== updating zmk-studio-messages (modules/msgs/zmk-studio-messages):
HEAD is now at 6cb4c28 refactor: Return detailed save changes response.

Your repo has been cloned to "/home/tom/repro/zmk-config-repro-attempt".

Here are some things you can do now:

 "zmk keyboard add"  Add a keyboard.
 "zmk code"          Edit the repo files.
 "zmk cd"            Move to the repo directory.
```

closed 2025-08-03 by tomchestna