zmkfirmware/zmk #2990: nice!nano v2 high power consumption after enabling `SOC_DCDC_NRF52X_HV` became default

Miigon, 2025-07-07:
commit f0b20c1c938da6dfd51f4639a2a71b587cab4205: feat(boards): Add nRF52 high voltage DC/DC config

```
feat(boards): Add nRF52 high voltage DC/DC config
Added a Kconfig option to enable SOC_DCDC_NRF52X_HV for nice_nano_v2
and mikoto. According to Nordic's documentation, the DC/DC regulator is
more efficient than the LDO regulator, so this is enabled by default.
```

Is this option actually tested?
I tested `SOC_DCDC_NRF52X_HV` on/off on both a genuine official nice!nano v2 and a nice!nano v2 aliexpress clone (dubbed `NRF52840 Development Board Compatible With Nice! Nano V2.0`).

The test is performed on a pair of corne split keyboard (no light, wireless split), with a multimeter in series with the battery connection.
Tested battery current draw on both standby (not in sleep mode, but not typing anything either. with active bluetooth connection to 1 computer), and pressed state(pressing a key without letting go continuously). Battery is at 8.2V, both board is tested on the same pair of battery, back to back.

The result showed a noticeable current draw increase with the option ON compared to OFF on the genuine nice!nano v2, with even more drastically increased current draw on the aliexpress nice!nano v2 clone board.

## Genuine nice!nano v2
shown an increase in current draw across the broad
|                | OFF    | ON     | diff    | diff% |
|----------------|--------|--------|---------|-------|
| Left(standby)  | 0.48mA | 0.60mA | +0.12mA | +25%  |
| Right(standby) | 0.05mA | 0.06mA | +0.01mA | +20%  |
| Left(pressed)  | 1.10mA | 1.58mA | +0.48mA | +44%  |
| Right(pressed) | 0.79mA | 1.22mA | +0.43mA | +54%  |

## aliexpress nice!nano v2 clone board
way worse. basically tanked the battery life:
|                | OFF    | ON     | diff    | diff%  |
|----------------|--------|--------|---------|--------|
| Left(standby)  | 0.60mA | 1.90mA | +1.30mA | +216%  |
| Right(standby) | 0.08mA | 1.40mA | +1.32mA | +1650% |
| Left(pressed)  | 1.37mA | 2.60mA | +2.23mA | +607%  |
| Right(pressed) | 1.01mA | 2.10mA | +1.09mA | +107%  |

Considering it's effect has been negative on both the genuine and the clone board(added here because the number of people who might've bought the clone board thinking it's the same as the genuine one might be significant), at least on my units. Maybe it's worthwhile to do more tests before enabling it by default?

----------------

For anyone who stumbles upon this issue trying to find a solution, you can add this option to your board's `.conf` file, compile and reflash it, and measure with a multimeter to see if it's the problem:
```
CONFIG_BOARD_ENABLE_DCDC_HV=n
```

joelspadin, 2025-07-12:
I was able to reproduce an increase in power usage, but not quite as large as you are seeing.

Test setup:

- nice!nano v2 with 2% milk shield
- Nordic PPK2 providing 4V power
- USB disconnected. BT connected to Windows 11 PC.

Results:

|               | DC-DC Off | DC-DC On | diff | diff% |
|----------|-------|------|---------|--------|
| Standby | 40 uA | 45 uA | 5 uA | +12.5% |
| Pressed | 0.94 mA | 1.17 mA | 0.23 mA | +24% |

I am surprised by these results. Either the DC-DC option makes more sense in different scenarios (higher load, lower operating voltage, etc.) or there's something wrong with Zephyr's implementation of it. We should probably change the default to disabled.

nmunnich, 2025-07-12:
Have any of you tested with a previous Zephyr version, e.g. https://github.com/zmkfirmware/zephyr/commit/489d3b00c49d1d3a0d4ec1f693fa0719db38bbf9? I'm wondering if something inside the nordic HAL might be messing with things somehow, and it was bumped Jun 16.

MickiusMousius, 2025-07-12:
Just tested then using the following commit: 

`commit 461f5c832fb8854d87dca54d113d306323697219 (HEAD)
Author: cormoran <cormoran707@gmail.com>
Date:   Sat May 10 09:01:17 2025 +0900

    fix(pointing): Avoids mutex leak for default layer toggle event (#2934)

    Remove unnecessary special check for toggle of layer 0,
    to avoid mutex leak.
`

In idle with DCDC I get 104uA, with LDO 103uA.

Switch held with DCDC I get 1.48mA, with LDO 1.49mA.

The variability of one run to the next is greater than the what I'm getting as the difference between LDO vs. DCDC.

Not much difference for me, kind of makes me thing I'd be better to simplify my PCB and skip 3 components.

MickiusMousius, 2025-07-12:
Oh dear, that is rather embarrassing. I’d forgotten that I’d need to checkout a different zephyr to get a different zephyr 🤦‍♂️

I shall do so in the morning. It’s 2223 here now.

MickiusMousius, 2025-07-13:
As it turns out I had not run "west update" in an age!

After west update I get the following:

* DCDC idle ~100uA, key held, 1.35mA
* LDO idle ~100uA, key held 1.38mA

Interestingly enough I've found the idle varies between 95uA & 105uA, however, in both test scenarios the key held had essentially no variability (10 second window used for averaging in NRF connect).

For me I'm willing to call the DCDC option more efficient. While the difference is repeatable it is very small, enough so to be essentially irrelevant.

Miigon, 2025-07-21:
just noticed I have incorrectly typed 8.2V as my battery voltage, fixed with edit, it's 4.2V.

PR #2995 (2025-07-21): Disable high voltage DC-DC by default
Given a typical setup of a battery at ~4 V and a nice!nano v2 running at 3.3 V, I measured a small increase in power usage (using a Nordic PPK2) when SOC_DCDC_NRF52X_HV was enabled. This changes the nice_nano_v2 and mikoto boards so this option is disabled by default.

Fixes #2990
commit 8059e671 (2025-07-23): fix(boards): Disable high voltage DC-DC by default (#2995)

Given a typical setup of a battery at ~4 V and a nice!nano v2 running at
3.3 V, I measured a small increase in power usage (using a Nordic PPK2)
when SOC_DCDC_NRF52X_HV was enabled. This changes the nice_nano_v2 and
mikoto boards so this option is disabled by default.

Fixes #2990
closed 2025-07-21 by nmunnich