zmkfirmware/zmk #2997: Persistent RGB power consumption until reset

MickiusMousius, 2025-07-12:
After deactivating the RGB on a board with a NRF52840 the controller continues to consumer an extra 1mA until the board is rebooted.

The board I've tested this on has a separate power control for the RGB so consumes less than 2uA when not in use.

I've verified that this isn't a pin leakage issue by setting the SPI pin for the LED strip to an unconnected pin on the controller, I'm still observing the same behaviour even though there is nothing attached to that pin.

MickiusMousius, 2025-07-13:
I've tested this with an oscilloscope today now, when the RGB is on there's a clear signal on the data pin for the RGB, when the RGB is off there's no signal.

This definitely looks as though the processor is wasting cycles on something.

afayge, 2025-08-01:
Yes if SPI3 is used as the signal output for RGB, it will be timed to produce additional CPU work (guessing, as there will be a fixed frequency producing power consumption at this point). change SPI3 to SPI2, Of course the circuit needs MOS to switch the power supply of the RGB, otherwise the RGB LED does not light up is also have and large current counts well.

afayge, 2025-08-06:
<yourboard>-pinctrl,dtsi, pinctrl changed all spi3_xxx to spi2_, 
and 
<yourboard>.dtsi   &spi3 to &spi2, 
pinctrl-0 = <&spi3_default>; and  pinctrl-1 = <&spi3_sleep>; 
to 
pinctrl-0 = <&spi2_default>; and  pinctrl-1 = <&spi2_sleep>;

maybe you can search all spi3 and replace to spi2 in your board files.

VeyPatch, 2025-08-27:
Wanted to add that I'm running in to the same issue (As discussed on discord: https://discord.com/channels/719497620560543766/1410186780585627688). Adding `CONFIG_ZMK_RGB_UNDERGLOW_ON_START=n` makes the power draw normal until you turn on the RGB LEDs and then back off again which makes it be in the off but consuming 1mA state.

VeyPatch, 2025-08-28:
Well the ZMK docs has a example implementation that uses spi3: https://zmk.dev/docs/development/hardware-integration/lighting/underglow. I think just changing this to spi2 and having a comment about adding SCK needing to be an unused pin could be enough for now.