zmkfirmware/zmk #3006: How do I build a config located in a sub folder ?

Nuclear-Squid, 2025-07-23:
This was probably answered somewhere else, but I could not find it. I’m sorry

Basically,I have a repo for my custom keyboard, containing everything I needed to develop the hardware. I’d like to have a `firmware` sub folder for the various firmwares I want to build, including ZMK. The problem is that the default build CI requires the repo to be the root repo of the config and doesn’t find it. How can I specify that my config is located in `firmware/zmk` and not at the root of the repo ?

Thanks

Nuclear-Squid, 2025-07-23:
I’ve gotten some progress ! A friend of mine showed me that some inputs could be overridden. So I added a couple lines to my build scrip, and ended up with this:

```yaml
on: [push, pull_request, workflow_dispatch]

jobs:
  build:
    uses: zmkfirmware/zmk/.github/workflows/build-user-config.yml@main
    with:
      build_matrix_path: firmware/zmk/build.yml
      config_path: firmware/zmk/config
```

Now the github action successfully finds the firmware folder, but seems to fail to update the `west workspace` it creates.

The workspace is initialized using `west init -l "/__w/Quacken/Quacken/firmware/zmk/config"` but updated using just `west update`, which produces the following error message:

```
FATAL ERROR: no west workspace found from "/__w/Quacken/Quacken"; "west update" requires one.
Things to try:
  - Change directory to somewhere inside a west workspace and retry.
  - Set ZEPHYR_BASE to a zephyr repository path in a west workspace.
  - Run "west init" to set up a workspace here.
  - Run "west init -h" for additional information.
```

Here are some links I should probably have given in the original post:

- Github repo (on my development branch): https://github.com/Nuclear-Squid/Quacken/tree/zmk_firmware
- Path to ZMK firmware folder: https://github.com/Nuclear-Squid/Quacken/tree/zmk_firmware/firmware/zmk
- Github action that fails to run: https://github.com/Nuclear-Squid/Quacken/actions/runs/16466961992/job/46546612464#step:9:20

If you guys know how to fix this, that would be pretty cool ^^

caksoylar, 2025-07-23:
Also see #2975

Nuclear-Squid, 2025-07-24:
Oh I did not see that, thanks ! It’s indeed pretty close to the changes I made, but cleaner / more robust. I’ll contribute patches there if I can meaningfully improve it ^^