zmkfirmware/zmk #3018: Feature Request: Update setup scripts for wired split support.

seapat, 2025-08-03:
Hi,

I am running the setup script [setup.ps1](https://github.com/zmkfirmware/zmk/blob/edf5c0814fd3ea202e43aad2d68fd32e882a518c/docs/src/templates/setup.ps1.mustache) via the command provided in the docs.
My keyboard / MCU choice (Sofle - 70 and Adafruit QT Py RP2040 - 2) lead to the script terminating with the following error: "Wired split is not yet supported by ZMK."

Since v0.3.0 added experimental support for wired split keyboards, the script should allow setting up with such a combination (Might be better to emit a warning instead).

Thanks!

petejohanson, 2025-08-03:
Unfortunately, pretty much all of the shields in our main ZMK repo use a single wire half-duplex setup for their wired split hardware, so can't be used today *anyways*. That setup script is also eventually going away in favor of ZMK CLI anywayse.

seapat, 2025-08-03:
I wasn't aware of the difference, thanks for the quick response!

closed 2025-08-03 by seapat