zmkfirmware/zmk #3025: West Build ignores Module Devicetree Overlays

r0b2g1t, 2025-08-05:
I try to migrate the deprecated approach of having the ```boards```-folder in the ```config```-folder of my ```zmk-config```-repo. But the West Build Github Action ignores the ```nice_view_adapter```-module which are detected:

```
-- Adding /__w/zmk-crkbd-config/zmk-crkbd-config/crkbd/boards/shields/crkbd
-- Adding /__w/zmk-crkbd-config/zmk-crkbd-config/crkbd/boards/shields/nice_view_adapter
-- Adding /__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/shields/nice_view_adapter
-- Adding /__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/shields/nice_view
```

A few steps further on, however, the default devicetrees from ```zmk```-folder are used:
```
-- Found BOARD.dts: /__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /__w/zmk-crkbd-config/zmk-crkbd-config/crkbd/boards/shields/crkbd/crkbd_left.overlay
-- Found devicetree overlay: /__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/shields/nice_view_adapter/nice_view_adapter.overlay
-- Found devicetree overlay: /__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/shields/nice_view_adapter/boards/nice_nano_v2.overlay
-- Found devicetree overlay: /__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/shields/nice_view/nice_view.overlay
-- Found devicetree overlay: /__w/zmk-crkbd-config/zmk-crkbd-config/config/crkbd.keymap
-- Found devicetree overlay: /__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/snippets/studio-rpc-usb-uart/studio-rpc-usb-uart.overlay
```

The shields are also not merged:
```
Parsing /__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/Kconfig
Loaded configuration '/__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/arm/nice_nano/nice_nano_v2_defconfig'
Merged configuration '/__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/prj.conf'
Merged configuration '/__w/zmk-crkbd-config/zmk-crkbd-config/config/crkbd.conf'
Merged configuration '/__w/zmk-crkbd-config/zmk-crkbd-config/crkbd/boards/shields/crkbd/crkbd_left.conf'
Merged configuration '/__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/shields/nice_view_adapter/nice_view_adapter.conf'
Merged configuration '/__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/boards/shields/nice_view/nice_view.conf'
Merged configuration '/__w/zmk-crkbd-config/zmk-crkbd-config/zmk/app/snippets/studio-rpc-usb-uart/studio-rpc-usb-uart.conf'
```

At first, I had the same problem with the ```corne``` module, but renaming it to ```crkbd``` worked for me. In general, however, the overlay approach does not work when customizing via a module.

The complete log-output:
[zmk-build-crkbd-left.log](https://github.com/user-attachments/files/21593565/zmk-build-crkbd-left.log)

nmunnich, 2025-08-05:
Hi, you should not be having shields in your local config that share a name with upstream shields. I do not believe we plan to change the build system to allow this. It increases confusion for anyone trying to help you with issues, or trying to use your work as a reference.

Depending on what you wanted to achieve by having modified copies of the upstream shields in your config there might be better solutions than merely renaming, though.

nmunnich, 2025-08-05:
To do that you shouldn't use the nice view adapter at all, but rather move the device tree nodes into your keyboard definition and adjust the pin definitions there.

Then you can build without the adapter being present at all.