zmkfirmware/zmk #3031: Studio and USB Logging conflict with rp2040

nmunnich, 2025-08-10:
#2900 has shown in the comments that there is a conflict present when both USB Logging and Studio are enabled. Studio can no longer connect to the device in this case.

petejohanson, 2025-08-10:
I wonder if logging enabled is blowing the RPC thread's stack? Could try bumping that for such a config and seeing if that helps.

peterjc, 2025-08-10:
I can run that test if you give something explicit for me to follow (i.e. what additional configuration lines and where to put it).

peterjc, 2025-08-11:
No joy with `CONFIG_ZMK_STUDIO_RPC_THREAD_STACK_SIZE=2048` or `CONFIG_ZMK_STUDIO_RPC_THREAD_STACK_SIZE=4048` using this branch https://github.com/peterjc/tutte-coxeter-zmk-config/tree/rpi-pico-rpc-test and GitHub Actions. Setting veified in the KConfig output:

```
CONFIG_ZMK_STUDIO_RPC=y
CONFIG_ZMK_STUDIO_RPC_RX_BUF_SIZE=30
CONFIG_ZMK_STUDIO_RPC_THREAD_STACK_SIZE=4048
CONFIG_ZMK_STUDIO_RPC_TX_BUF_SIZE=64
```

To be clear, the firmware builds, and the tester appears in the ZMK Studio connection screen (twice - separate macOS quirk), but cannot be connected.

Disabling USB logging and rebuilding fixes the ZMK Studio connection.